m5stack/Core2-for-AWS-IoT-Kit #6: Creating cert cannot just leave all values blank

gherlein, 2020-12-22:
"A window will open asking for additional information to include into the certificate. For purposes of this tutorial, we will leave all fields blank."

- it's not a window, it's in the terminal
- something needs to be there - otherwise it errors out

147dda0c29e5:esp_cryptoauth_utility gherlein$ openssl req -new -x509 -key output_files/signerkey.pem -out output_files/signercert.crt -days 365
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) []:
State or Province Name (full name) []:
Locality Name (eg, city) []:
Organization Name (eg, company) []:
Organizational Unit Name (eg, section) []:
Common Name (eg, fully qualified host name) []:
Email Address []:
error, no objects specified in config file
problems making Certificate Request

gherlein, 2020-12-22:
Works if you just enter a two digit country code

rashedtalukder, 2020-12-22:
Might depend on your version of OpenSSL. Our testing across multiple users and versions did not show this issue. However, I've updated: https://edukit.workshop.aws/en_uk/blinky-hello-world/device-provisioning.html

rashedtalukder, 2020-12-22:
https://edukit.workshop.aws/en_uk/blinky-hello-world/device-provisioning.html

closed 2020-12-22 by rashedtalukder