m5stack/Core2-for-AWS-IoT-Kit #8: instructions for finding serial number are wrong

gherlein, 2020-12-22:
https://edukit.workshop.aws/en/blinky-hello-world/device-provisioning.html

"For optional fleet management, you’ll need to replace the common name value with the device’s serial number. You can retrieve the device’s serial number by going to the Core2-for-AWS-IoT-EduKit-Factory-Firmware folder and entering the command idf.py build flash monitor -p <<DEVICE_PORT>>. Replace «DEVICE_PORT» with the system port your device is logically mounted on."

idf.py not found.  Also, it's a nit, but the device is not logically mounted - better to say "Replace «DEVICE_PORT» with the serial port device your Core2 is connected to"

rashedtalukder, 2020-12-22:
If you've followed the earlier pre-req for the tutorial on https://edukit.workshop.aws/en/blinky-hello-world/prerequisites.html#installing-esp-idf-v42, the ESP-IDF. will be added to your path. You'll need to run the **export.sh** (Linux/OSX) or **export.bat** (Windows) script to add it back to your path again. Additionally, you'll need to reactivate your conda environment if it isn't already.

They are logically mounted, but they are not a logical volume since there's none. I agree that it would be simpler to write it that way.

rashedtalukder, 2020-12-22:
Thank you for the great feedback @gherlein. Updated the verbiage to be simpler: https://edukit.workshop.aws/en/blinky-hello-world/device-provisioning.html#generating-a-device-certificate-manifest

Hopefully running the ESP-IDF's export script resolved the "idf.py not found" issue for you.

closed 2020-12-22 by rashedtalukder