m5stack/Core2-for-AWS-IoT-Kit #9: Blinky-Hello-World: Device provisioning fails with Python x509 module exception

karbak, 2020-12-23:
During manifest generation as part of the device provisioning process in the Blinky-Hello-World tutorial, the following command:
```
python secure_cert_mfg.py --flash sample_bins/secure_cert_mfg_esp32.bin --signer-cert output_files/signercert.crt --signer-cert-private-key output_files/signerkey.pem -p <<DEVICE_PORT>>
```
, errors out as follows:
```
Validate Device Certificate:
OK

<device-cert-id>
reading slot 0 public key
>>

reading slot 1 public key
>>

reading slot 2 public key
>>

reading slot 3 public key
>>

reading slot 4 public key
>>

Traceback (most recent call last):
  File "/Users/<user>/projects/esp/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/components/esp-cryptoauthlib/esp_cryptoauth_utility/secure_cert_mfg.py", line 192, in <module>
    main()
  File "/Users/<user>/projects/esp/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/components/esp-cryptoauthlib/esp_cryptoauth_utility/secure_cert_mfg.py", line 97, in main
    hs.manifest.generate_manifest_file(esp, args, init_mfg)
  File "/Users/<user>/projects/esp/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/components/esp-cryptoauthlib/esp_cryptoauth_utility/helper_scripts/manifest.py", line 295, in generate_manifest_file
    log_key_id = jws_b64encode(log_cert.extensions.get_extension_for_class(x509.SubjectKeyIdentifier).value.digest)
  File "/Users/<user>/.espressif/python_env/idf4.2_py3.9_env/lib/python3.9/site-packages/cryptography/x509/extensions.py", line 135, in get_extension_for_class
    raise ExtensionNotFound(
cryptography.x509.extensions.ExtensionNotFound: No <class 'cryptography.x509.extensions.SubjectKeyIdentifier'> extension was found
```

This is on MacOS Catalina, running inside a Python 3.7.9 virtualenv, having followed all of the prerequisite steps.

karbak, 2020-12-24:
It turns out I had a different Python 3.7.9 being picked up in my shell path from the Conda - switching to the Conda-installed 3.7.9 Python and correct virtualenv fixed this issue. Please feel free to close it - it may be worth re-iterating on the device provisioning page that the cryptography library from a non-Conda Python 3.7.9 installation doesn't seem to work for this process. Thanks, and apologies for the noise.

rashedtalukder, 2020-12-24:
As long as your conda environment is activated, your "python" PATH variable should reference the conda Python 3.7.9 installation regardless of what your base system PATH point to.

rashedtalukder, 2020-12-27:
Closing since we haven't had an update.

closed 2020-12-27 by rashedtalukder