m5stack/Core2-for-AWS-IoT-Kit #10: Upload and Monitor via Pio

francescom1973, 2020-12-24:
I am getting the following error (username blanked with xxxxxx) when running Upload and Monitor. 

File "c:\users\xxxxxxxx\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/cu.SLAB_USBtoUART': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)

I am on Windows 10 and the Kit COM port is COM9. How do I commit the port to the configuration - it does not seem to save correctly. 
Thanks.

francescom1973, 2020-12-24:
I resolved the issue above. However now when I run Build or Upload and Monitor I get the following: 

Error: Couldn't find target config target-AWS_IoT_EduKit-Getting_Started.elf-c69afd479f49d5de2667.json

francescom1973, 2020-12-24:
I resolved the issue above by configuring the COM port correctly with the format COM[9]. 

Might be worth documenting these errors in the documentation in case others come across same ?

rashedtalukder, 2020-12-24:
I believe your underlying request is to move the cumbersome steps for Windows to be located within the documentation itself, rather than pointing to Espressif's documentation or PlatformIO's. If you look at the section for identifying the port, it calls out what you need to do for the upload_port option in Windows: https://edukit.workshop.aws/en/getting-started/run-rainmaker.html#identifying-and-setting-the-serial-port-on-host-machine

I will evaluate this—especially if others feel the same way. Leaving the issue open for now.

closed 2020-12-30 by rashedtalukder