m5stack/Core2-for-AWS-IoT-Kit #12: AWS IoT EduKit > Blinky Hello World > Device Provisioning - OpenSSL is missing

i40playbook, 2020-12-27:
I have started the example on a new VM with Windows 10 and it looks like OpenSSL needs to be installed first before running the example:

C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\sample_certs>openssl req -new -x509 -key output_files/signerkey.pem -out output_files/signercert.crt -days 365
Can't open C:\Program Files\Common Files\ssl/openssl.cnf for reading, No such file or directory
3568:error:02001003:system library:fopen:No such process:crypto\bio\bss_file.c:69:fopen('C:\Program Files\Common Files\ssl/openssl.cnf','r')
3568:error:2006D080:BIO routines:BIO_new_file:no such file:crypto\bio\bss_file.c:76:

rashedtalukder, 2020-12-28:
I will have this fixed tomorrow.

rashedtalukder, 2020-12-28:
Fixed with the commit below. Device provisioning is now executed via a Python script using PyOpenSSL and uses OpenSSL provided by the miniconda installer. Tested on a VM with vanilla Windows 10 - version 1909 & version 20H2, OSX Big Sur, OSX High Sierra, and Ubuntu 18.04.

https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/8f8d4d791ef4c69c2d9b1cc0569c9733f7ef2b7c

rashedtalukder, 2020-12-28:
Fixed with the commit https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/8f8d4d791ef4c69c2d9b1cc0569c9733f7ef2b7c. Device provisioning is now executed via a Python script using PyOpenSSL and uses OpenSSL provided by the miniconda installer. Tested on a VM with vanilla Windows 10 - version 1909 & version 20H2, OSX Big Sur, OSX High Sierra, and Ubuntu 18.04.

closed 2020-12-28 by rashedtalukder