m5stack/Core2-for-AWS-IoT-Kit #13: AWS IoT EduKit > Blinky Hello World > Device Provisioning - esptool is missing

i40playbook, 2020-12-27:
After I´ve installed OpenSLL and copied openssl.cnf to ...\commen files\ssl the step "openssl req -new -x509 -key output_files/signerkey.pem -out output_files/signercert.crt -days 365" was possible and I have entered the certificate values.

But step "python secure_cert_mfg.py --flash sample_bins/secure_cert_mfg_esp32.bin --signer-cert output_files/signercert.crt --signer-cert-private-key output_files/signerkey.pem -p <<DEVICE_PORT>>" with DEVICE_PORT = COM3 failed (see at the end "ModuleNotFoundError: No module named 'esptool"):

(edukit) C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility>python secure_cert_mfg.py --flash sample_bins/secure_cert_mfg_esp32.bin --signer-cert output_files/signercert.crt --signer-cert-private-key output_files/signerkey.pem -p COM3
Traceback (most recent call last):
  File "secure_cert_mfg.py", line 17, in <module>
    import helper_scripts as hs
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\__init__.py", line 2, in <module>
    from .serial import load_app_stub, cmd_interpreter, esp_cmd_check_ok
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\serial.py", line 22, in <module>
    import esptool
**ModuleNotFoundError: No module named 'esptool'**

rashedtalukder, 2020-12-27:
Have you run the export script for esp-idf after restarting? Since you're on Windows, it will be "\path\to\esp\esp-idf\**export.bat**"

Might be a good option to create a new shortcut that opens command prompt (or anaconda prompt), activates your edukit environment, and runs the esp-idf export script.

i40playbook, 2020-12-27:
Thank you much for the feedback ! 
I´ve ran the export script (a second time) - and I got:

(edukit) C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility>python secure_cert_mfg.py --flash sample_bins/secure_cert_mfg_esp32.bin --signer-cert output_files/signercert.crt --signer-cert-private-key output_files/signerkey.pem -p COM3
Pyserial is not installed for C:\Users\info\.conda\envs\edukit\python.exe. Check the README for installation instructions.
Traceback (most recent call last):
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\serial.py", line 22, in <module>
    import esptool
ModuleNotFoundError: No module named 'esptool'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "secure_cert_mfg.py", line 17, in <module>
    import helper_scripts as hs
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\__init__.py", line 2, in <module>
    from .serial import load_app_stub, cmd_interpreter, esp_cmd_check_ok
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\serial.py", line 28, in <module>
    import esptool
  File "C:\Users\info\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 38, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

i40playbook, 2020-12-27:
ESP-IDF folder is: C:\Users\info\esp\esp-idf (using the default path caused an error during installation "path to long")

rashedtalukder, 2020-12-27:
Do you see esptool if you enter **echo %Path%** ?

i40playbook, 2020-12-27:
(edukit) C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility>echo %Path%
C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\info\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\info\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\cmake\3.16.4\bin;C:\Users\info\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\info\.espressif\tools\ninja\1.10.0\;C:\Users\info\.espressif\tools\idf-exe\1.0.1\;C:\Users\info\.espressif\tools\ccache\3.7\;C:\Users\info\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\info\esp\esp-idf\tools;C:\Users\info\.conda\envs\edukit;C:\Users\info\.conda\envs\edukit\Library\mingw-w64\bin;C:\Users\info\.conda\envs\edukit\Library\usr\bin;C:\Users\info\.conda\envs\edukit\Library\bin;C:\Users\info\.conda\envs\edukit\Scripts;C:\Users\info\.conda\envs\edukit\bin;C:\ProgramData\Miniconda3\condabin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2;C:\Users\info\AppData\Local\Microsoft\WindowsApps;C:\Users\info\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\info\AppData\Local\GitHubDesktop\bin

rashedtalukder, 2020-12-27:
It's not listed in your path. The export script didn't return any errors?

i40playbook, 2020-12-27:
(edukit) C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility>%userprofile%\esp\esp-idf\**export.bat**
Setting IDF_PATH: C:\Users\info\esp\esp-idf

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\info\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\info\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\cmake\3.16.4\bin;C:\Users\info\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\info\.espressif\tools\ninja\1.10.0\;C:\Users\info\.espressif\tools\idf-exe\1.0.1\;C:\Users\info\.espressif\tools\ccache\3.7\;C:\Users\info\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\info\esp\esp-idf\tools;C:\Users\info\.conda\envs\edukit;C:\Users\info\.conda\envs\edukit\Library\mingw-w64\bin;C:\Users\info\.conda\envs\edukit\Library\usr\bin;C:\Users\info\.conda\envs\edukit\Library\bin;C:\Users\info\.conda\envs\edukit\Scripts;C:\Users\info\.conda\envs\edukit\bin;C:\ProgramData\Miniconda3\condabin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2;C:\Users\info\AppData\Local\Microsoft\WindowsApps;C:\Users\info\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\info\AppData\Local\GitHubDesktop\bin

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pyserial>=3.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
esp-windows-curses; sys_platform == 'win32'
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: C:\Users\info\.conda\envs\edukit\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\info\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\info\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\cmake\3.16.4\bin;C:\Users\info\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\info\.espressif\tools\ninja\1.10.0\;C:\Users\info\.espressif\tools\idf-exe\1.0.1\;C:\Users\info\.espressif\tools\ccache\3.7\;C:\Users\info\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\info\esp\esp-idf\tools;C:\Users\info\.conda\envs\edukit;C:\Users\info\.conda\envs\edukit\Library\mingw-w64\bin;C:\Users\info\.conda\envs\edukit\Library\usr\bin;C:\Users\info\.conda\envs\edukit\Library\bin;C:\Users\info\.conda\envs\edukit\Scripts;C:\Users\info\.conda\envs\edukit\bin;C:\ProgramData\Miniconda3\condabin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2;C:\Users\info\AppData\Local\Microsoft\WindowsApps;C:\Users\info\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\info\AppData\Local\GitHubDesktop\bin

i40playbook, 2020-12-27:
Looks like some thing is wrong in Python?

rashedtalukder, 2020-12-27:
Looks like the dependencies weren't installed and other configurations not set. Have you run the installer batch file:

https://edukit.workshop.aws/en/blinky-hello-world/prerequisites.html#installing-esp-idf-v42 Windows (64-bit) → Step 6

i40playbook, 2020-12-27:
I´ve tried to run the export.bat on C:\ ... let me check again

(edukit) C:\>%userprofile%\esp\esp-idf\export.bat
Setting IDF_PATH: C:\Users\info\esp\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pyserial>=3.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
esp-windows-curses; sys_platform == 'win32'
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: C:\Users\info\.conda\envs\edukit\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;: C:\Users\info\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\info\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\info\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\info\.espressif\tools\cmake\3.16.4\bin;C:\Users\info\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\info\.espressif\tools\ninja\1.10.0\;C:\Users\info\.espressif\tools\idf-exe\1.0.1\;C:\Users\info\.espressif\tools\ccache\3.7\;C:\Users\info\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\info\esp\esp-idf\tools;C:\Users\info\.conda\envs\edukit;C:\Users\info\.conda\envs\edukit\Library\mingw-w64\bin;C:\Users\info\.conda\envs\edukit\Library\usr\bin;C:\Users\info\.conda\envs\edukit\Library\bin;C:\Users\info\.conda\envs\edukit\Scripts;C:\Users\info\.conda\envs\edukit\bin;C:\ProgramData\Miniconda3\condabin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2;C:\Users\info\AppData\Local\Microsoft\WindowsApps;C:\Users\info\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\info\AppData\Local\GitHubDesktop\bin

(edukit) C:\>

i40playbook, 2020-12-27:
still the same error message - I´ll try your suggestion

(edukit) C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility>python secure_cert_mfg.py --flash sample_bins/secure_cert_mfg_esp32.bin --signer-cert output_files/signercert.crt --signer-cert-private-key output_files/signerkey.pem -p COM3
Pyserial is not installed for C:\Users\info\.conda\envs\edukit\python.exe. Check the README for installation instructions.
Traceback (most recent call last):
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\serial.py", line 22, in <module>
    import esptool
ModuleNotFoundError: No module named 'esptool'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "secure_cert_mfg.py", line 17, in <module>
    import helper_scripts as hs
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\__init__.py", line 2, in <module>
    from .serial import load_app_stub, cmd_interpreter, esp_cmd_check_ok
  File "C:\IoT\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\serial.py", line 28, in <module>
    import esptool
  File "C:\Users\info\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 38, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

rashedtalukder, 2020-12-27:
No, the **import**.bat

i40playbook, 2020-12-27:
Sorry - Yes, I´ve ran it - let me try again

i40playbook, 2020-12-27:
(edukit) C:\Users\info\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Users\info\.espressif\python_env\idf4.2_py3.7_env
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\info\\.conda\\envs\\edukit\\Lib\\venv\\scripts\\nt\\python.exe'
Traceback (most recent call last):
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1208, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\info\.conda\envs\edukit\lib\subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\info\\.conda\\envs\\edukit\\python.exe', '-m', 'virtualenv', 'C:\\Users\\info\\.espressif\\python_env\\idf4.2_py3.7_env']' returned non-zero exit status 1.

i40playbook, 2020-12-27:
Do you mean import.bat or install.bat (see "6. Run the ESP-IDF installation script to ensure everything is set correctly: install.bat")?

rashedtalukder, 2020-12-27:
Sorry, brain is sleepy. Early AM here. Meant install.bat

So to help me replicate your issue, you used the installer tool at: https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe
Or did you clone the ESP-IDF directly and tried manually installing everything?

i40playbook, 2020-12-27:
No problem, I appreciate your help! 

Yes, I have used the "esp-idf-tools-setup-2.3.exe" as Admin and I have chosen "release 4.2" for install. The installation failed at the very final step (see log below) with 
"2020-12-26 19:17:10.133   Failed to remove temporary directory: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp"

2020-12-26 19:10:44.049   Log opened. (Time zone: UTC+01:00)
2020-12-26 19:10:44.049   Setup version: Inno Setup version 6.0.2 (u)
2020-12-26 19:10:44.049   Original Setup EXE: C:\Users\info\Downloads\esp-idf-tools-setup-2.3.exe
2020-12-26 19:10:44.049   Setup command line: /SL5="$5041A,350580993,721408,C:\Users\info\Downloads\esp-idf-tools-setup-2.3.exe" 
2020-12-26 19:10:44.049   Windows version: 10.0.19042  (NT platform: Yes)
2020-12-26 19:10:44.049   64-bit Windows: Yes
2020-12-26 19:10:44.049   Processor architecture: x64
2020-12-26 19:10:44.049   User privileges: None
2020-12-26 19:10:44.049   Administrative install mode: No
2020-12-26 19:10:44.049   Install mode root key: HKEY_CURRENT_USER
2020-12-26 19:10:44.049   64-bit install mode: Yes
2020-12-26 19:10:44.065   Created temporary directory: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp
2020-12-26 19:10:44.065   -- DLL function import --
2020-12-26 19:10:44.065   Function name: idpAddFile
2020-12-26 19:10:44.065   DLL name: files:idp.dll
2020-12-26 19:10:44.065   Extracting temporary file: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.081   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.081   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpAddFileComp
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpAddMirror
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpAddFtpDir
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpAddFtpDirComp
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpClearFiles
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpFilesCount
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpFtpDirsCount
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
2020-12-26 19:10:44.113   Importing the DLL function.
2020-12-26 19:10:44.113   Successfully imported the DLL function. Delay loaded? No
2020-12-26 19:10:44.113   -- DLL function import --
2020-12-26 19:10:44.113   Function name: idpFileDownloaded
2020-12-26 19:10:44.113   DLL name: files:idp.dll
2020-12-26 19:10:44.113   Dest DLL name: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp\idp.dll
[... 580 lines cut ...]
                              main(sys.argv[1:])
                            File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1488, in main
                              action_func(args)
                            File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1181, in action_install
                              tool_obj.download(tool_version)
                            File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 598, in download
                              raise DownloadError()
                          __main__.DownloadError
                          
2020-12-26 19:15:30.551   --------
2020-12-26 19:15:30.551   Message box (OK):
                          Installation has failed with exit code 1
2020-12-26 19:17:03.560   User chose OK.
[... cut ...]
                          Display the installation log now?
2020-12-26 19:17:09.865   User chose Yes.
2020-12-26 19:17:09.901   CurStepChanged raised an exception.
2020-12-26 19:17:10.059   Need to restart Windows? No
2020-12-26 19:17:10.093   Exception message:
2020-12-26 19:17:10.093   Message box (OK):
                          Out Of Range.
2020-12-26 19:17:10.125   User chose OK.
2020-12-26 19:17:10.125   Deinitializing Setup.
2020-12-26 19:17:10.133   Failed to remove temporary directory: C:\Users\info\AppData\Local\Temp\is-2BLP8.tmp

rashedtalukder, 2020-12-27:
Are you able to download the compiler zip file manually? 
https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip

I can retrieve that file no problem on my side.

i40playbook, 2020-12-27:
Yes - I have been able to download it manually (slower than it should be) - next step ;-) ?

i40playbook, 2020-12-27:
If I try to copy the ZIP file to C:\Users\info\.espressif\dist = error is "already there"

i40playbook, 2020-12-27:
running install.bat again 

(edukit) C:\Users\info\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Users\info\.espressif\python_env\idf4.2_py3.7_env
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\info\\.conda\\envs\\edukit\\Lib\\venv\\scripts\\nt\\python.exe'
Traceback (most recent call last):
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1208, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\info\.conda\envs\edukit\lib\subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\info\\.conda\\envs\\edukit\\python.exe', '-m', 'virtualenv', 'C:\\Users\\info\\.espressif\\python_env\\idf4.2_py3.7_env']' returned non-zero exit status 1.

rashedtalukder, 2020-12-27:
Try deactivating your conda environment for a second by running "conda deactivate"

i40playbook, 2020-12-27:
Running in base

(base) C:\Users\info\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Users\info\.espressif\python_env\idf4.2_py3.8_env
Installing virtualenv
Collecting virtualenv
  Using cached virtualenv-20.2.2-py2.py3-none-any.whl (5.7 MB)
Requirement already satisfied: six<2,>=1.9.0 in c:\programdata\miniconda3\lib\site-packages (from virtualenv) (1.15.0)
Collecting filelock<4,>=3.0.0
  Using cached filelock-3.0.12-py3-none-any.whl (7.6 kB)
Collecting distlib<1,>=0.3.1
  Using cached distlib-0.3.1-py2.py3-none-any.whl (335 kB)
Collecting appdirs<2,>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Installing collected packages: filelock, distlib, appdirs, virtualenv
  WARNING: The script virtualenv.exe is installed in 'C:\Users\info\AppData\Roaming\Python\Python38\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed appdirs-1.4.4 distlib-0.3.1 filelock-3.0.12 virtualenv-20.2.2
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\ProgramData\\Miniconda3\\Lib\\venv\\scripts\\nt\\python.exe'
Traceback (most recent call last):
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "C:\Users\info\esp\esp-idf\tools\idf_tools.py", line 1207, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
  File "C:\ProgramData\Miniconda3\lib\subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\ProgramData\\Miniconda3\\python.exe', '-m', 'virtualenv', 'C:\\Users\\info\\.espressif\\python_env\\idf4.2_py3.8_env']' returned non-zero exit status 1.

i40playbook, 2020-12-27:
As I´m working on a win10-VM (VMware Workstation 16) I could go back and try from scratch or after installing Miniconda again?

rashedtalukder, 2020-12-27:
I don't think we'll get a resolution from that. We tested this on Windows before launch without these issues. You're having issues related to Espressif's software (ESP-IDF) and rather than steer you towards them, I'd like to try and find a solution that will help. I'll need time to debug and test on my VM.

i40playbook, 2020-12-27:
OK, great - I look forward on your solution. Even if the AWS Edukit description is already a very detailed step by step manual it seems to miss some steps if you start on an empty VM (like OpenSSL)? I´ve continued with the Jupyter notebook part and to honested if you are not familiar with Jupyter notebooks this section is challenging as well ;-)
My goal is to clarify whether we could use the AWS Edukit as standard example in our IoT training - currently I would say there is room for improvement to make it more stable and convient to ensure the training success ;-)

rashedtalukder, 2020-12-27:
I agree about the Jupyter section and we will be removing it. You'll actually see significant updates coming VERY shortly to the entire thing registration section.

i40playbook, 2020-12-27:
By the way: the MS VSCode and Platformio example was very smooth and easy to accomplish (and thank you very much for the tipp to erase the MCU if the WIFI credentials are wrong ;-)

rashedtalukder, 2020-12-27:
That's great to hear! We are keen to see ESP-IDF v4.2 support in PlatformIO and offer that as an option in the future.

We certainly would love to have this used in your IoT training. Keep providing feedback and we'll work on resolving the issues you see.

i40playbook, 2020-12-27:
OK, as I´m working with M5Stack hardware for some years now I was very curios to combine this with AWS IoT and I´ve ordered a Core2AWS right away for Christmas ... maybe I should wait a little ... instead of being the very early adopter ;-)

rashedtalukder, 2020-12-27:
Hopefully the direct line of support and quick turnaround on fixes is a better than other early adoption experiences.

i40playbook, 2020-12-27:
Direct line of support on a Sunday is impressive - and I look forward to this co-operation and solution to make this Edukit successfull! Making IIoT available for a broader range of engineers or even business people would be a great benefit...

rashedtalukder, 2020-12-27:
I am able to replicate the issue and it looks like there's an issue with the ESP-IDF Tools installer itself. Specifically the python script below. Digging into it.

 C:\Users\XX\miniconda3\python.exe "C:\Users\XX\Desktop\esp-idf\tools\idf_tools.py" --idf-path C:\Users\XX\Desktop\esp-idf install

rashedtalukder, 2020-12-27:
Solution was to switch the order of ESP-IDF installation with Miniconda. The automated installer seems to get confused and doesn't end up using PyOpenSSL to be able to make https requests.

rashedtalukder, 2020-12-29:
@i40playbook let me know if you've had a chance to try out the new set of instructions.

closed 2020-12-30 by rashedtalukder