m5stack/Core2-for-AWS-IoT-Kit #14: Issue installing Silicon Labs USB to UART bridge driver

emcfins, 2020-12-28:
Not a bug per se, but thought this would be helpful to other people.

I was getting an error when trying to install the driver: 
```
Result: err:(null) errResult:Can't open CFBundle for /Library/Extensions/SiLabsUSBDriver.kext. Can't create /Library/Extensions/SiLabsUSBDriver.kext.
```

I followed the guidance [here](https://medium.com/@robin.grosset/fixed-cp210x-driver-not-working-on-macos-high-sierra-a758a5d7704a) and was able to install the new driver successfully.

Basically, I needed to install the legacy drivers first. That prompted me to allow the install of the drivers from the Silicon Labs Inc developer, which was blocked by my system.
After allowing the developer, the legacy install completed and then the install of the updated drivers completed with no errors.

rashedtalukder, 2020-12-28:
Thank you for reporting and sharing the fix for others. I've updated the installation instructions to give macOS 10.13+ specific instructions for allowing developers through macOS Gatekeeper. I couldn't replicate your particular need to install the legacy drivers first, but I'm only able to test against macOS 10.14, 10.15, and 11.

emcfins, 2020-12-29:
It's possible that the process of installing the legacy drivers simply gave me the ability to see the error message and to prompt me to allow the developers in Gatekeeper, I didn't try it again without installing them.

closed 2020-12-30 by rashedtalukder