m5stack/Core2-for-AWS-IoT-Kit #15: AWS IoT EduKit > Getting Started > Running the ESP RainMaker Agent - Error claiming with Rainmaker app

emcfins, 2020-12-28:
I've restarted the device - both with the reload button as well as turning it off and on - and retried the claiming a few times, all with the same issue.
I updated my phone app and have tried to do the claiming through my browser as well. All with the same error.

Claiming finished with error: 
Claim init failed

Claiming failed. Please reboot the device and restart provisioning.

emcfins, 2020-12-28:
[image]

rashedtalukder, 2020-12-28:
Sorry you're having issues. What is the output in your serial monitor? That would give better insights on what's going wrong on the device side.

Have you also tried [erasing the firmware](https://edukit.workshop.aws/en/getting-started/run-rainmaker.html#erasing-the-firmware-with-platformio) and then running "flash and monitor" again to start on a empty firmware?

emcfins, 2020-12-28:
Hi @rashedtalukder - thanks for the suggestions. 

It was clear to me that it was my phone app. I ended up completely deleting and re-installing the app on my phone and it worked.

To answer your question, the only message I was seeing in the serial monitor was that provisioning was starting

```
[0;32mI (4102) app_wifi: Provisioning Started. Name : PROV_42a64e, POP : 66d2a6ff␛[0m
␛[0;32mI (20932) protocomm_nimble: mtu update event; conn_handle=0 cid=4 mtu=185
```

Reinstalling resolved my issue. Thank you!

closed 2020-12-28 by emcfins