m5stack/Core2-for-AWS-IoT-Kit #19: Error installing windows Curses

scarolan, 2020-12-30:
This happens during the "Build" step of Getting Started, on a Windows 10 machine:

```
Installing windows-curses package
Processing c:\users\sean\.platformio\packages\framework-espidf\tools\kconfig_new\esp-windows-curses
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: 'C:\\Users\\sean\\.platformio\\packages\\framework-espidf\\tools\\kconfig_new\\esp-windows-curses'
```

scarolan, 2020-12-30:
Also seems cmake and gcc are missing:

```
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: 'C:\\Users\\sean\\.platformio\\packages\\framework-espidf\\tools\\kconfig_new\\esp-windows-curses'

*** Error 1
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "C:/Users/sean/Documents/PlatformIO/Projects/M5-Stack-Core2-for-AWS/.pio/build/core2foraws/CMakeFiles/CMakeOutput.log".
See also "C:/Users/sean/Documents/PlatformIO/Projects/M5-Stack-Core2-for-AWS/.pio/build/core2foraws/CMakeFiles/CMakeError.log". 

CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    C:\Users\sean\.platformio\packages\framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:6 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:6 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
  ```

rashedtalukder, 2020-12-30:
Sorry you're having this strange issue. Quick search netted me this on PIO's community forums: https://community.platformio.org/t/cant-create-esp-idf-project-correctly-in-platformio/16370/17

I tested and found that this issue only arises when you already have the ESP-IDF installed on your machine and then trying to install PIO. The esp-window-curses was introduced in ESP-IDF v4.1. Hopefully doing the steps listed in the forum post helps. I recommend adding your issue to PIO forums or their github issues. I will reach out to PlatformIO and see if there's a simpler solution for everyone.

Closing because this is not an issue with provided software or content or the reference hardware.

closed 2020-12-30 by rashedtalukder

scarolan, 2020-12-30:
@rashedtalukder you might wish to add a warning to your quickstart for Windows users who already have Platform IO installed.  I'll try the steps on the expressif forums and see if that resolves the issues.

rashedtalukder, 2020-12-30:
The issue arises if you have the ESP-IDF installed already, not PlatformIO. I added the warning yesterday.

fervenceslau, 2022-06-15:
Why did it not work when installing Espressif 32 through the embedded tab of platformio? It installed a package of version 4.0.1 and it expected a version 4.4.1... A friend of mine solved this by telling me to add `platform = https://github.com/platformio/platform-espressif32.git` the .ini file.