m5stack/Core2-for-AWS-IoT-Kit #20: Download of ESP-IDF tools is painfully slow

scarolan, 2020-12-30:
You might wish to host these files on a CDN or even on github.  Or maybe AWS!

I'm about five hours into the installation process and it still has a long way to go.  The download speed is measured in kilobits per second.

https://edukit.workshop.aws/en/blinky-hello-world/prerequisites.html#installing-esp-idf-v42

rashedtalukder, 2020-12-30:
The ESP-IDF is on GitHub. The background process is retrieving the file from GitHub. AWS or M5Stack does not own the ESP-IDF.

closed 2020-12-30 by rashedtalukder

hamrik, 2021-01-04:
The IDF libraries are on GitHub indeed, but the build tools are not. The compilers that are downloaded by the `install.sh` or `install.bat` scripts for example are pulled from [dl.espressif.com](dl.espressif.com) which limits download speeds to 80 kbps.