m5stack/Core2-for-AWS-IoT-Kit #21: Windows installation requires pip

scarolan, 2020-12-30:
pip should be bundled in the installer script that installs python.

Here is the fix, after I did this the ESP-IDF tools completed installing.

download https://bootstrap.pypa.io/get-pip.py
 C:\Python39\python.exe .\get-pip.py

and if the ESP-IDF tools is upstream, maybe just add a note in the docs for Windows users to install pip manually.

rashedtalukder, 2020-12-30:
Pip is installed with the Python installer during ESP-IDF installer run and also with conda. Something is wrong on you particular machine or you bypassed those installations some how. This is abnormal.

scarolan, 2020-12-30:
Sorry I didn't realize that you actually have to install two completely separate Python development environments.  That's a pretty heavy lift just to program an IoT microcontroller.  I've got miniconda installed now, closing this issue.

closed 2020-12-30 by scarolan

rashedtalukder, 2020-12-30:
You don't _need_ to. The problem is is that during testing we found so many issues (like the ones you've been having) with conflicts that it provided a significantly better experience to isolate using Conda. One of the big pitfalls really is that the Windows ESP-IDF installer forces a Python installation but also barks at miniconda. We would prefer to install miniconda first and rely solely on the virtual environment created, but had to switch due to https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/issues/13. I will raise this issue with Espressif and see what they say.

The issues that arise with Python version conflicts is not unique to IoT development.