m5stack/Core2-for-AWS-IoT-Kit #23: syntax error in registration_helper.py

scarolan, 2020-12-30:
Script output:

```
C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\utilities\AWS_IoT_registration_helper>python registration_helper.py -p COM6
Traceback (most recent call last):
  File "C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\utilities\AWS_IoT_registration_helper\registration_helper.py", line 53, in <module>
    import helper_scripts as esp_hs
  File "C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\__init__.py", line 4, in <module>
    from .manifest import generate_manifest_file
  File "C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\components\esp-cryptoauthlib\esp_cryptoauth_utility\helper_scripts\manifest.py", line 47, in <module>
    from jose import utils
  File "C:\Users\sean\.espressif\python_env\idf4.2_py3.9_env\lib\site-packages\jose.py", line 546
    print decrypt(deserialize_compact(jwt), {'k':key},
          ^
SyntaxError: invalid syntax
```

rashedtalukder, 2020-12-30:
Please use Conda and be sure to install the necessary dependencies and their versions for the registration helper by running:

pip3 install -r requirements.txt

scarolan, 2020-12-30:
I'm using conda now, but not without having to install some extra modules.

I will open a new issue for the new errors in conda.

closed 2020-12-30 by scarolan