m5stack/Core2-for-AWS-IoT-Kit #25: Blinky-Hello-World> Connnecting to AWS IoT Core > Not seeing AWS IoT EduKit Configuration in KConfig

kevkearney, 2020-12-30:
[image]

rashedtalukder, 2020-12-30:
Hi, please verify you are in the `/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World` directory.

kevkearney, 2020-12-30:
I am.

kevkearney, 2020-12-30:
I was able to set the AWS Iot Endpoint Hostname, but I can't find the next part, for SSID and wifi key.

kevkearney, 2020-12-30:
I also don't see it in the Kconfig file in the project.

kevkearney, 2020-12-30:
Ah wait, I see it in kconfig.projbuild. Let me see if I can get it to build.

kevkearney, 2020-12-30:
Sorry, let me be more clear:
I see the kconfig options in the main folder, in kconfig.projbuild, but for some reason when I run the `idf.py menuconfig` command, they aren't included in the menu.

rashedtalukder, 2020-12-30:
The idf.py menuconfig command reads the kconfig files from each of the linked dependencies. In this case:
`/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/main/Kconfig.projbuild`

For some reason it's not reading it for you. Can you list your OS and version as well? Also run a `idf.py fullclean` to erase any strange build configurations. Even if you modify your kconfig.projbuild to have your Wi-Fi credentials, when you begin the build process it with generate a **sdkconfig** file, which contains all of your application configurations and is what the ESP Tools then parse for your linker and compile options.

kevkearney, 2020-12-30:
Sure, let me provide more info:

Ubuntu 20.04 on a clean install specific for the purposes of playing with the m5stack

Here is a pastebin of my output:
https://pastebin.com/DkCB55c8

rashedtalukder, 2020-12-30:
I am setting up a fresh test environment to see if I can duplicate, just that it takes a while :)

kevkearney, 2020-12-30:
No problem, I'm trying to figure it out on my side as well.
Let me know where I can be of assistance.

rashedtalukder, 2020-12-30:
Resolved in https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/aa99a24f8708c9a65abf1066ec14460bba56a915. Filename case was incorrect (needs capital **K**). Menuconfig never threw any errors so we didn't catch it. Espressif's documentation also has the wrong file name in their docs (https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/aa99a24f8708c9a65abf1066ec14460bba56a915). Thanks for filing this issue!

closed 2020-12-30 by rashedtalukder

kevkearney, 2020-12-30:
Yup, it worked. Thanks