m5stack/Core2-for-AWS-IoT-Kit #26: registration_helper.py is missing esptool and pyserial

scarolan, 2020-12-30:
Ran this inside the miniconda environment as documented here. This was after running `pip3 install -r requirements.txt` which worked fine.

https://edukit.workshop.aws/en/blinky-hello-world/device-provisioning.html

```
python registration_helper.py -p COM3
```

```
Pyserial is not installed for C:\Users\sean\miniconda3\envs\edukit\python.exe. Check the README for installation instructions.
Traceback (most recent call last):
  File "C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\utilities\AWS_IoT_registration_helper\registration_helper.py", line 42, in <module>
    import esptool
ModuleNotFoundError: No module named 'esptool'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\sean\git_repos\Core2-for-AWS-IoT-EduKit\Blinky-Hello-World\utilities\AWS_IoT_registration_helper\registration_helper.py", line 49, in <module>
    import esptool
  File "C:\Users\sean\Desktop\esp-idf-2\components\esptool_py\esptool\esptool.py", line 38, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
```

scarolan, 2020-12-30:
The fix:

```
python3 -m pip install esptool pyserial
```

Note that `pip3 install esptool pyserial` did *not* work. I think it was finding the pip from the *other* development environment and installing the modules there.  The command above worked for me on Windows 10.

rashedtalukder, 2020-12-30:
esptool and pyserial is installed via the Windows 10 ESP-IDF installer. It looks like it's most likely not added to path if you already have installed it. You need to run `%userprofile%\Desktop\esp-idf\export.bat` each time you restart your **Anaconda Prompt**. Optionally you can update your shortcut to activate your **edukit** conda environment and execute the export script each time you open Anaconda Prompt.

scarolan, 2020-12-30:
Thanks @rashedtalukder, I have everything working now!  Appreciate your help.

closed 2020-12-30 by scarolan

rashedtalukder, 2020-12-30:
I am evaluating your feedback overall and evaluating other options for Windows users. Maybe Windows Subsystem for Linux would provided an easier path forward.

scarolan, 2020-12-30:
Yes, I'd recommend WSL; it will probably be easier for you to support.  I will try setting this up on my other machine which has WSL2 installed.  

I imagine the only tricky bit will be the serial port; am not sure whether WSL can properly expose the serial port to PlatformIO.