m5stack/Core2-for-AWS-IoT-Kit #28: Hello World - Blinking the LEDs fails

scarolan, 2020-12-30:
I have my device successfully connecting to AWS IoT:

```
I (10500) aws_iot: Attempting to use device certificate from ATECC608
I (16800) Blinky: Successfully connected to AWS IoT Core!
I (16800) Blinky: Subscribing...
I (17040) Blinky: Subscribed to topic: 012333b76cac4c3701/#
I (17040) Blinky: 
I (17040) Blinky: **************************************
I (17040) Blinky: Client Id - 012333b76cac4c3701
I (17050) Blinky: **************************************
```

I subscribed to the # topic per the instructions, but when I publish nothing happens on the device:

[image]

scarolan, 2020-12-30:
Never mind, i was in the wrong region!

closed 2020-12-30 by scarolan

rashedtalukder, 2020-12-30:
Thank you for reporting, Sean. I've updated all links that point to the console to go to us-west-2 so anyone in the future doesn't encounter the same issue.

muriloplinio, 2021-06-14:
I have the same problem and this post helps me!! Thank you!

wrong region!

Now it is working!