m5stack/Core2-for-AWS-IoT-Kit #30: AWS CLI VERSION

kspray, 2020-12-31:
This is my first time posting an issue in github. My apologies if the format isn't correct. The [instructions ](https://edukit.workshop.aws/en/blinky-hello-world/prerequisites.html) for the Blinky Hello World example walks you through installing AWS CLI Version 2, but I realized that the requirements.txt file found in the "Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/utilities/AWS_IoT_registration_helper" folder actually has it installing AWS CLI version "awscli>=1.18.202" is this correct?

Actually I was really confused as the instructions for this have changed in the past week. It had you installing the miniconda and creating an environment and using Jupyter notebook to edit and run a file "TNGTLS_aws_connect", but it appears that whole process was replaced by this registration helper.py file. 

One other issue I noticed was that If you download the Core2-for-aws-iot-edukit and extract it from github, you get messages saying that some of the filestrings are too long for the windows/dos environment and can't be installed. 
Thanks

rashedtalukder, 2020-12-31:
Hi, yes the requirements file lists 1.18 and up. AWS CLI 2.x satisfies this requirement without throwing an error for individuals who might already have the CLI installed and for whatever reason do not want to upgrade. The APIs used do not have a conflict so we've chosen to not cause an issue where there wouldn't be one. 

We removed the Jupyter steps since people found it cumbersome and without value. We only included it since the utility provided by Microchip was in Jupyter for some reason. Would it be helpful to have a notice for when the content/repo has been updated?

Thank you for letting me know about the file length issue in Windows. We will evaluate cloning from C:\ or something to that effect.

kspray, 2020-12-31:
Ok Thanks! Yes, maybe if the instructions were dated or had a revision date or something, it would be helpful indeed.

rashedtalukder, 2021-01-03:
Added revision date to the repo link on edukit.workshop.aws: https://edukit.workshop.aws/

closed 2021-01-03 by rashedtalukder