m5stack/Core2-for-AWS-IoT-Kit #32: queue.c:1148 (xQueueGiveFromISR)- assert failed

sbstjn, 2021-01-01:
Hi, I recently received the hardware device and tried https://edukit.workshop.aws/. The "Getting Started" example worked fine, but the "Blinky Hello World" example does not work.

After running `idf.py flash monitor` the devices is stuck and fails with:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7364
load:0x40078000,len:14980
ho 0 tail 12 room 4
load:0x40080400,len:4388
0x40080400: _init at ??:?

entry 0x40080694
I (31) boot: ESP-IDF v4.3-dev-2136-gb0150615d 2nd stage bootloader
I (31) boot: compile time 22:57:31
I (31) boot: chip revision: 3
I (35) qio_mode: Enabling default flash chip QIO
I (41) boot.esp32: SPI Speed      : 80MHz
I (45) boot.esp32: SPI Mode       : QIO
I (50) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f7a00
I (93) boot:  3 ota_0            OTA app          00 10 00410000 00010000
I (101) boot:  4 ota_1            OTA app          00 11 00420000 00640000
I (108) boot:  5 spiffs           Unknown data     01 82 00a60000 004c4c00
I (116) boot: End of partition table
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=22fe8h (143336) map
I (171) esp_image: segment 1: paddr=00033010 vaddr=3ffbdb60 size=04b64h ( 19300) load
I (177) esp_image: segment 2: paddr=00037b7c vaddr=40080000 size=00404h (  1028) load
I (178) esp_image: segment 3: paddr=00037f88 vaddr=40080404 size=08090h ( 32912) load
I (197) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=ba59ch (763292) map
I (422) esp_image: segment 5: paddr=000fa5c4 vaddr=40088494 size=0c6c4h ( 50884) load
I (452) boot: Loaded app from partition at offset 0x10000
I (452) boot: Disabling RNG early entropy source...
I (463) psram: This chip is ESP32-D0WD
I (464) spiram: Found 64MBit SPI RAM device
I (464) spiram: SPI RAM mode: flash 80m sram 80m
I (469) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (476) cpu_start: Pro cpu up.
I (480) cpu_start: Starting app cpu, entry point is 0x40081550
0x40081550: call_start_cpu1 at /Users/sbstjn/Applications/esp-idf/components/esp_system/port/cpu_start.c:133

I (471) cpu_start: App cpu up.
I (980) spiram: SPI SRAM memory test OK
I (988) cpu_start: Pro cpu start user code
I (988) cpu_start: cpu freq: 240000000
I (988) cpu_start: Application information:
I (991) cpu_start: Project name:     AWS_IoT_EduKit-Blinky-Hello-Wor
I (998) cpu_start: App version:      162805a
I (1003) cpu_start: Compile time:     Jan  1 2021 22:58:10
I (1009) cpu_start: ELF file SHA256:  6ce88d6ebc1a0f76...
I (1015) cpu_start: ESP-IDF:          v4.3-dev-2136-gb0150615d
I (1021) heap_init: Initializing. RAM available for dynamic allocation:
I (1029) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (1035) heap_init: At 3FFCC448 len 00013BB8 (78 KiB): DRAM
I (1041) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1048) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1054) heap_init: At 40094B58 len 0000B4A8 (45 KiB): IRAM
I (1060) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1069) spi_flash: detected chip: generic
I (1073) spi_flash: flash io: qio
I (1078) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1088) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1118) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3
I (1328) ILI9341: Initialization.
I (1528) ILI9341: Display orientation: LANDSCAPE
I (1528) ILI9341: 0x36 command value: 0x08
I (1568) wifi:wifi driver task: 3ffd3720, prio:23, stack:6656, core=0
I (1568) system_api: Base MAC address is not set
I (1568) syst/Users/sbstjn/Applications/esp-idf/components/freertos/queue.c:1148 (xQueueGiveFromISR)- assert failed!

abort() was called at PC 0x4009039a on core 0
0x4009039a: xQueueGiveFromISR at /Users/sbstjn/Applications/esp-idf/components/freertos/queue.c:1148 (discriminator 3)

Backtrace:0x4008c547:0x3ffbe920 0x4008cd4d:0x3ffbe940 0x40093422:0x3ffbe960 0x4009039a:0x3ffbe9d0 0x40085f82:0x3ffbe9f0 0x4008a1bf:0x3ffbea20 0x4008a319:0x3ffbea40 0x40084669:0x3ffbea70 0x400d3ec9:0x3ffd30f0 0x400d4bd9:0x3ffd3110 0x400d3631:0x3ffd3130 0x4000bd83:0x3ffd3150 0x4000117d:0x3ffd3170 0x400592fe:0x3ffd3190 0x4005937a:0x3ffd31b0 0x40058bbf:0x3ffd31d0 0x4017a48b:0x3ffd3200 0x4017fa6f:0x3ffd3230 0x4017fbf1:0x3ffd3540 0x4018a5b1:0x3ffd3570 0x400932b5:0x3ffd35a0 0x400d9fc3:0x3ffd35f0 0x4014c48f:0x3ffd3620 0x4014998e:0x3ffd3640 0x401636b6:0x3ffd3660 0x4008f869:0x3ffd3690
0x4008c547: panic_abort at /Users/sbstjn/Applications/esp-idf/components/esp_system/panic.c:367

0x4008cd4d: esp_system_abort at /Users/sbstjn/Applications/esp-idf/components/esp_system/system_api.c:108

0x40093422: abort at /Users/sbstjn/Applications/esp-idf/components/newlib/abort.c:46

0x4009039a: xQueueGiveFromISR at /Users/sbstjn/Applications/esp-idf/components/freertos/queue.c:1148 (discriminator 3)

0x40085f82: spi_ready at /Users/sbstjn/Workspace/src/github.com/sbstjn/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/components/core2forAWS/tft/disp_spi.c:176

0x4008a1bf: spi_post_trans at /Users/sbstjn/Applications/esp-idf/components/driver/spi_master.c:581 (discriminator 1)

0x4008a319: spi_intr at /Users/sbstjn/Applications/esp-idf/components/driver/spi_master.c:614

0x40084669: _xt_lowint1 at /Users/sbstjn/Applications/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400d3ec9: uart_tx_char at /Users/sbstjn/Applications/esp-idf/components/vfs/vfs_uart.c:159

0x400d4bd9: uart_write at /Users/sbstjn/Applications/esp-idf/components/vfs/vfs_uart.c:217

0x400d3631: esp_vfs_write at /Users/sbstjn/Applications/esp-idf/components/vfs/vfs.c:423 (discriminator 4)

0x4017a48b: __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:403

0x4017fa6f: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)

0x4017fbf1: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x4018a5b1: esp_log_writev at /Users/sbstjn/Applications/esp-idf/components/log/log.c:189

0x400932b5: esp_log_write at /Users/sbstjn/Applications/esp-idf/components/log/log.c:199

0x400d9fc3: esp_read_mac at /Users/sbstjn/Applications/esp-idf/components/esp_common/src/mac_addr.c:148 (discriminator 13)

0x4014c48f: wifi_init_process at ??:?

0x4014998e: ieee80211_ioctl_process at ??:?

0x401636b6: ppTask at ??:?

0x4008f869: vPortTaskWrapper at /Users/sbstjn/Applications/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 6ce88d6ebc1a0f76

Rebooting...

```

rashedtalukder, 2021-01-02:
Did you erase the flash after finishing the getting started example?

sbstjn, 2021-01-02:
I did run `idf.py erase_flash`, yes.

I tried a second hardware device without going through the "Getting Started" example. It's the same result.

rashedtalukder, 2021-01-02:
I am looking into it. It looks like Espressif updated v4.2 (or some of the dependencies) and this has caused a break. I will let you know as soon as it's fixed.

rashedtalukder, 2021-01-02:
You are on the master branch or v4.3 of the ESP-IDF --> `I (31) boot: ESP-IDF v4.3-dev-2136-gb0150615d 2nd stage bootloader`. 4.3 is not a stable release and we do not support it.

Please follow the instructions to install v4.2: https://edukit.workshop.aws/en/blinky-hello-world/prerequisites.html#installing-esp-idf-v42

closed 2021-01-02 by rashedtalukder

sbstjn, 2021-01-03:
With `4.2` any command with `idf.py` fails with `-- Configuring incomplete, errors occurred!`

I cannot event run menuconfig with 4.2

sbstjn, 2021-01-03:
I wiped everything, now I can at least built the blinking lights example. Thanks for the help! 🙌

rashedtalukder, 2021-01-03:
Maybe I should've added more detail. To go to 4.2 from 4.3, you'll need to remove the esp-idf directory _and_ the $HOME/.espressif directory. Then recursively clone esp-idf 4.2, run the install script, and then the export script.