m5stack/Core2-for-AWS-IoT-Kit #36: The guiTask sometimes stops.

arms22, 2021-01-06:
The guiTask does not get out of the while loop of lv_refr_vdb_flush.
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/d5b92eff833e6fc4cbefa23114646214dc055fb9/Smart-Thermostat/components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c#L759

In disp_spi_transaction, I think it is because the call to spi_device_queue_trans returns ESP_ERR_NO_MEM and does not start SPI transfer.
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/d5b92eff833e6fc4cbefa23114646214dc055fb9/Smart-Thermostat/components/core2forAWS/tft/disp_spi.c#L141

rashedtalukder, 2021-01-06:
Thank you for reporting this. I'll take a look today.

rashedtalukder, 2021-01-07:
Approximately how long do you leave it running before you see this error?

arms22, 2021-01-07:
Occurs frequently while using WiFi and Bluetooth.
It seems to occur easily when transferring data of 40960bytes (DISP_BUF_SIZE * sizeof(lv_color_t)).

rashedtalukder, 2021-01-08:
Ok I see, so I'm assuming this is a modified project and not the standard Smart Thermostat project that we provide (since we don't use Bluetooth).

So what is happening is that you're passing in too much data to the display library to be shown. The 40960 byte limit is the buffer limit for the display and is calculating by:

Display width  x  64 bytes (number of pixels)  x  2 bytes (16 bit color depth):
320px * 64 * 2 = 40960bytes

The flush callback copies the buffer's content to the display, but your buffer is already full, so it can't. Solution is to shrink down whatever it is you're trying to send to be 320px*240px (or 320px*320px and the bottom 80px will not show up) so that it fits. https://docs.lvgl.io/v7/en/html/porting/display.html#display-driver

Is that helpful?

arms22, 2021-01-08:
Yes, I'm modifying the Smart Thermostat project.
I am sorry that I did not tell that to you.

I changed the DISP_BUF_SIZE to the following and the problem seems to be gone.

#define DISP_BUF_SIZE (LV_HOR_RES_MAX * 32)

Thank you.

closed 2021-01-08 by arms22

rashedtalukder, 2021-01-08:
Well, that is one way...but you're just wasting memory and who knows what other consequences of trying to write to an area that doesn't exist. Can you reduce the size of whatever you're trying to display?