m5stack/Core2-for-AWS-IoT-Kit #37: Can't install hidapi on MacOS Catalina

dustnic, 2021-01-06:
I cannot install the requirements for the AWS_IoT_Registration_helper on MacOS Catalina.
```
/usr/bin/xcodebuild -version 
Xcode 12.3
Build version 12C33

sw_vers 
ProductName:	Mac OS X
ProductVersion:	10.15.7
BuildVersion:	19H114
```

```ERROR: Command errored out with exit status 1:
   command: /Users/narnoldi/.pyenv/versions/3.8.1/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/tmp/pip-install-pc_4srqk/hidapi/setup.py'"'"'; __file__='"'"'/private/tmp/pip-install-pc_4srqk/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/tmp/pip-wheel-3l5sdfn5
       cwd: /private/tmp/pip-install-pc_4srqk/hidapi/
  Complete output (2185 lines):
  running bdist_wheel
  running build
  running build_ext
  skipping 'hid.c' Cython extension (up-to-date)
  building 'hid' extension
  creating build
  creating build/temp.macosx-10.14-x86_64-3.8
  creating build/temp.macosx-10.14-x86_64-3.8/hidapi
  creating build/temp.macosx-10.14-x86_64-3.8/hidapi/mac
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -framework IOKit -framework CoreFoundation -framework AppKit -Ihidapi/hidapi -I/Users/narnoldi/.pyenv/versions/3.8.1/include/python3.8 -c hid.c -o build/temp.macosx-10.14-x86_64-3.8/hid.o
  clang: warning: -framework IOKit: 'linker' input unused [-Wunused-command-line-argument]
  clang: warning: -framework CoreFoundation: 'linker' input unused [-Wunused-command-line-argument]
  clang: warning: -framework AppKit: 'linker' input unused [-Wunused-command-line-argument]
  In file included from hid.c:14:
  In file included from /Users/narnoldi/.pyenv/versions/3.8.1/include/python3.8/Python.h:25:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:64:
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:93:16: warning: pointer is missing a nullability type specifier (_Nonnull, _Nullable, or _Null_unspecified) [-Wnullability-completeness]
          unsigned char   *_base;
                          ^
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:93:16: note: insert '_Nullable' if the pointer may be null
          unsigned char   *_base;
                          ^
                            _Nullable
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:93:16: note: insert '_Nonnull' if the pointer should never be null
          unsigned char   *_base;
                          ^
                            _Nonnull
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:138:32: warning: pointer is missing a nullability type specifier (_Nonnull, _Nullable, or _Null_unspecified) [-Wnullability-completeness]
          int     (* _Nullable _read) (void *, char *, int);
                                            ^
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:138:32: note: insert '_Nullable' if the pointer may be null
          int     (* _Nullable _read) (void *, char *, int);
                                            ^
                                             _Nullable
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:138:32: note: insert '_Nonnull' if the pointer should never be null
          int     (* _Nullable _read) (void *, char *, int);
                                            ^
                                             _Nonnull
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:138:40: warning: pointer is missing a nullability type specifier (_Nonnull, _Nullable, or _Null_unspecified) [-Wnullability-completeness]
          int     (* _Nullable _read) (void *, char *, int);
                                                    ^
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:138:40: note: insert '_Nullable' if the pointer may be null
          int     (* _Nullable _read) (void *, char *, int);
                                                    ^
   ```
(truncated)

rashedtalukder, 2021-01-06:
You’re on Python 3.8 and not using conda/miniconda. Please follow the prescriptive instructions to remove variability like what you’re experiencing.

dustnic, 2021-01-06:
installing with conda fixed the issue

closed 2021-01-06 by dustnic