m5stack/Core2-for-AWS-IoT-Kit #39: Intro to Alexa fot IoT

shurillu, 2021-01-16:
Hello,

   I'm trying to run the example for Alexa, but I'm stuck in the Amazon authentication phase.
Here the detail:

1. **Flash the certificates stage**: all seems to go fine (I got the certificates by email, I modified the CSV file, I generated the mfg.bin file and I loaded the certificate into my Core2)
2. **Flash the firmware stage**: all seems to go fine. The Core2 boots, side leds blinks red, the display shows "ALEXA APP" and the slider to set the screen brightness.
3. **Smartphone Espressif "ESP Alexa" App stage**: I downloaded the Alexa App for iPhone (iOS ver 14.3). Running the app, I tap on "Add New Device" button and the pone found my Core2 as "ESP-Alexa-A7FC". Tapping on it, the app show the "Proof of Possession" code (abcd1234), tapping on "Done" button the app show an "amazon alexa" message and ask for signing in with Amazon.
4. When I tap on that button (sign in with Amazon) in the serial log I see these messages:
```
I (732594) [avs_config]: Received request for device details
I (732594) [avs_config]: code_verifier - abcd1234, code_challenge - 6c7nGrky_ehjM40Ivk3p3-OeoEm9r7NCzmWexUULaa4
I (732594) [avs_config]: version: 1, DSN: b8f009c4a7fc, ProductID: Core2ForAWS

```
My phone shows this:
[image]

The italian message says something like that: _We apologies. An error has occurred while we are trying to manage your request._

At this point I can't do anything else.

If I skip the Amazon sign in phase, I can set the WiFi credentials but without an account, the Core2 can't login with Amazon: the Core 2 side leds are blue pulsing cyan.  This is the serial log output

```
W (1702644) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

I (1725394) [app_prov]: Received Wi-Fi credentials:
        SSID: mySSID
        Password: myPASSWORD
I (1727844) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1728544) wifi:state: init -> auth (b0)
I (1728554) wifi:state: auth -> assoc (0)
I (1728554) wifi:state: assoc -> run (10)
I (1728594) wifi:connected with mySSID, aid = 19, channel 6, BW20, bssid = 00:1c:82:05:2c:58
I (1728604) wifi:security: WPA2-PSK, phy: bgn, rssi: -34
I (1728604) wifi:pm start, type: 1

I (1728714) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[app_wifi]: Connected with IP Address: 192.168.1.65
I (1729454) esp_netif_handlers: sta ip: 192.168.1.65, mask: 255.255.255.0, gw: 192.168.1.1
I (1729454) wifi_prov_mgr: STA Got IP
I (1729454) [app_prov]: Provisioning successful
I (1732724) wifi_prov_mgr: Provisioning stopped
I (1732724) [app_prov]: Provisioning stopped
I (1732724) wifi_prov_scheme_ble: BTDM memory released
I (1732734) [va_nvs_utils]: No value set for: friendlyname
I (1732734) [va_nvs_utils]: No value set for: udn
I (1732744) SSDP: Network Interface List (1):
[alexa]: Waiting for time to be updated
I (1732744) SSDP: 1. wifi  : 192.168.1.65
I (1732764) LSSDP: create SSDP socket 58
[alexa]: Done getting current time: 1610793030
I (1734064) [tls_certification]: Done setting global CA store
I (1734064) [va_nvs_utils]: No value set for: clientId
I (1734064) [va_nvs_utils]: No value set for: refreshToken
I (1734074) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.

```
The last message is repeated.

Any clues?

Thanks, 

Stefano

nishit-jain-esp, 2021-01-16:
@shurillu : Have you by any chance modified the Product ID to Core2ForAWS, since thats what is showing in the log. Can you please confirm if in the file `Core2-for-AWS-IoT-EduKit/Alexa_for_IoT-Intro/main/Kconfig.projbuild` the value for ALEXA_PRODUCT_ID is set to `esp_alexa_open`

shurillu, 2021-01-16:
Hello nishit-jain-esp
   thank you for your response. I can confirm that I used the source code downloaded yesterday from github, master branch.
I checked the `Kconfig.projbuild` and the `ALEXA_PRODUCT_ID` is set to `esp_alexa_open`. Here the snipped extracted from the `Kconfig.projbuild` file:

```
config ALEXA_PRODUCT_ID
    string "Device Product ID"
    default "esp_alexa_open"
    help
        Product ID that will be used by the device to connect with AVS.
```

Hoping that could be useful to find the issue, Here the full log from the boot to the moment when the Amazon logging stage abort.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7488
ho 0 tail 12 room 4
load:0x40078000,len:14368
load:0x40080400,len:5360
entry 0x40080710
I (31) boot: ESP-IDF v4.2-218-g8911c2912 2nd stage bootloader
I (31) boot: compile time 18:10:02
I (31) boot: chip revision: 3
I (35) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) qio_mode: Enabling default flash chip QIO
I (47) boot.esp32: SPI Speed      : 80MHz
I (52) boot.esp32: SPI Mode       : QIO
I (57) boot.esp32: SPI Flash Size : 16MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 fctry_aia        WiFi data        01 02 00010000 00006000
I (100) boot:  3 fctry            WiFi data        01 02 00016000 00006000
I (107) boot:  4 otadata          OTA data         01 00 0001c000 00002000
I (115) boot:  5 ota_0            OTA app          00 10 00020000 003f0000
I (123) boot:  6 ota_1            OTA app          00 11 00410000 003f0000
I (130) boot: End of partition table
I (134) boot_comm: chip revision: 3, min. application chip revision: 0
I (142) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1afe1c (1768988) map
I (736) esp_image: segment 1: paddr=0x001cfe44 vaddr=0x3ffbdb60 size=0x001d4 (   468) load
I (736) esp_image: segment 2: paddr=0x001d0020 vaddr=0x400d0020 size=0x13e994 (1304980) map
I (1174) esp_image: segment 3: paddr=0x0030e9bc vaddr=0x3ffbdd34 size=0x0493c ( 18748) load
I (1182) esp_image: segment 4: paddr=0x00313300 vaddr=0x40080000 size=0x00404 (  1028) load
I (1182) esp_image: segment 5: paddr=0x0031370c vaddr=0x40080404 size=0x1da9c (121500) load
I (1255) boot: Loaded app from partition at offset 0x20000
I (1255) boot: Disabling RNG early entropy source...
I (1256) psram: This chip is ESP32-D0WD
I (1260) spiram: Found 64MBit SPI RAM device
I (1265) spiram: SPI RAM mode: flash 80m sram 80m
I (1270) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1278) cpu_start: Pro cpu up.
I (1281) cpu_start: Application information:
I (1286) cpu_start: Project name:     AWS_IoT_EduKit-Intro_to_Alexa
I (1293) cpu_start: App version:      1
I (1298) cpu_start: Compile time:     Jan 15 2021 18:09:38
I (1304) cpu_start: ELF file SHA256:  8f09fec98a3c2837...
I (1310) cpu_start: ESP-IDF:          v4.2-218-g8911c2912
I (1316) cpu_start: Starting app cpu, entry point is 0x40082104
I (0) cpu_start: App cpu up.
I (1825) spiram: SPI SRAM memory test OK
I (1829) heap_init: Initializing. RAM available for dynamic allocation:
[... 118 lines cut ...]
I (5124) [audio_codec]: Codec mp3_decoder Event Started
[basic_player]: Codec event: Codec Started
I (5124) MP3Decoder: Found ID3 tag, skipping 45 bytes
I (5134) MP3Decoder: MP3 sync operation...
I (5134) MP3Decoder: MP3 sync success
I (5144) [audio_codec]: Codec mp3_decoder Event Set Freq
[basic_player]: Codec event: Codec Set Frequency
I (5154) [basic_player]: Set Freq event: 24000, 1, 16
I (5164) [prompt]: Processing request ===> Pipeline Started
I (8394) MP3Decoder: Codec reader finished
I (8704) MP3Decoder: Done with last chunk
I (8704) MP3Decoder: Closed
I (8704) [audio_codec]: Codec mp3_decoder Event Stopped
[basic_player]: Codec event: Codec Stopped
I (9564) [prompt]: Processing request ===> Pipeline Stopped
I (9564) [focus_manager]: Release channel: Prompt Prompt
I (9564) [focus_manager]: Releasing Prompt
I (9574) [sys_playback]: Release
I (44184) [avs_config]: Received request for device details
I (44184) [avs_config]: code_verifier - abcd1234, code_challenge - 6c7nGrky_ehjM40Ivk3p3-OeoEm9r7NCzmWexUULaa4
I (44194) [avs_config]: version: 1, DSN: b8f009c4a7fc, ProductID: Core2ForAWS
```

Thanks,
Stefano

rashedtalukder, 2021-01-17:
Hi Stefano, could you please do a `git pull` and grab the latest commit? The sdkconfig.defaults was changed and the product name was changed in there. I’ve reverted it. 

Alternatively, you can run `idf.py menuconfig`, go to voice assistant configurations and change the product ID to `esp_alexa_open`

shurillu, 2021-01-17:
Hello rashedtalukder
   I downloaded the latest commit and it works perfectly! Thank you so much!

Stefano

closed 2021-01-17 by shurillu