m5stack/Core2-for-AWS-IoT-Kit #40: Documentation Addition (AWS IoT EduKit > Smart Thermostat > Data acquisition)

MKPeak, 2021-01-18:
I just noticed that the Unix steps are included but not the steps for Windows.

> Data Acquisition...
> ...Code sample...
> ...2.Change directory to the **Smart-Thermostat** project:
> ```cd Core2ForAWS-AWS-IoT-EduKit/Smart-Thermostat```
> 3. Copy your ```sdkconfig``` file from the **Blinky-Hello-World** demo to simplify setting your Wi-Fi and AWS endpoint:
> ```cp ../Blinky-Hello-World/sdkconfig .```

May want to include a note for Windows users that a different process is required. Instead they should be in the Blinky-Hello-World folder and use ```copy sdkconfig ..\Smart-Thermostat``` to copy ```sdkconfig``` and then go to the Smart-Thermostat folder.

rashedtalukder, 2021-01-20:
Thank you! I see the issue. I will test the changes and update in the next 24hrs.

rashedtalukder, 2021-01-21:
Updated. Thanks for reporting @MKPeak !

closed 2021-01-21 by rashedtalukder