m5stack/Core2-for-AWS-IoT-Kit #45: Could not build getting started on Ubuntu ARM64 Raspberry Pi

bathtimefish, 2021-02-23:
I tried to run getting started on Raspberry Pi Ubuntu 20.10 Desktop ARM64

```
uname -a
Linux user-desktop 5.8.0-1015-raspi #18-Ubuntu SMP PREEMPT Fri Feb 5 06:09:58 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

I got an error when run build process as bellow.

```
Processing core2foraws (platform: https://github.com/platformio/platform-espressif32.git; framework: espidf; board: esp32dev)
-------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80400.0
Error: Could not find the package with 'platformio/toolchain-xtensa32 @ ~2.80400.0' requirements for your system 'linux_aarch64'
The terminal process "pio 'run', '--environment', 'core2foraws'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

It seems that the toolchain can not be available for ARM64.
Does someone  knows any workaround?

brianivie, 2021-02-24:
This thread seems to indicate there might be one: https://github.com/platformio/platform-espressif32/issues/236
based on the workaround for the issue found here: https://github.com/esphome/issues/issues/594

rashedtalukder, 2021-02-25:
If you can avoid it, I would recommend not running it on a Raspberry Pi. The performance will be so limited compared to a laptop in the past several years that compiling each of the binaries will probably take hours.

That said, this seems [PlatformIO feature request](https://github.com/platformio/platform-espressif32/issues). Let us know if the suggestion by Brianivie works for you. We'll think about testing and incorporating this workflow if there is enough demand, but my assumption is that the experience will be extremely poor.

bathtimefish, 2021-02-28:
@brianivie Thank you for your advice. However I could not solve the probrem to tried those workarounds.
I guess its platformio problem that Tool Manager always failed to download in spite of installed new toolchan in ~/.platformio/pacakges/ .

I make sence that its difficult to setup esp32 dev environment on arm64 computer for now.

bathtimefish, 2021-02-28:
@rashedtalukder I tried it but could not works fine for now. I understood to difficult to use RPi for getting start Edukit thanks  your advice.

I throught that if I can prepare dev environment on RPi, I can increase easily the environment for education by copied SSD. However I may shoud give up it because it will be unconfortable if compiling  was slow.

closed 2021-03-05 by rashedtalukder