m5stack/Core2-for-AWS-IoT-Kit #47: ESP Alexa iOS app crashing on iPhone11 Pro Max

vbaranek, 2021-03-03:
ESP Alexa iOS app crashing on iPhone11 Pro Max (fresh install):
1) ALEXA App is running on M5 waiting for app registration (I can see screen brightness slider on screen and red light is flashing on the side)
2) ESP Alexa App is up and running on IPhone
3) I press button "Add New Device" on iPhone
4)My M5 is listed in devices via Bluetooth as "ESP-Alexa-7700"
5) Selected device in list
6) Amazon Alexa screen appear with dailog "Proof of possession" and "abcd1234" in text field, press button "done"
7) ESP Alexa app disappear from iPhone screen...

Any idea what to do?

vikas-chandra-mnnit, 2021-03-03:
@vbaranek I have tried above steps on iPhone 11 version 14.3, unfortunately I am unable to reproduce the crash. Kindly provide the below information to help us identify the issue, thanks.
1. iOS version.
2. Frequency of crash.

rashedtalukder, 2021-05-14:
Closing since there hasn't been any response.

closed 2021-05-14 by rashedtalukder