m5stack/Core2-for-AWS-IoT-Kit #51: Smart Thermostat - Device program constantly resetting after publishing through AWS (and lights not working)

David-Mesa, 2021-04-08:
Hello,

I went through this exercise multiple times and a couple of unexpected errors keep occurring. 

In step **c) Data Sync**, there is a bug during the validation step. Once a shadow message is published, the M5Core 2 device starts resetting the program and attempting reconnection (connection with AWS is stable before publishing the message).

If the validation is skipped without publishing, the device and monitoring keep working correctly. Then, a similar error occurs building the cloud application, once AWS IoT starts publishing messages to the device. The device keeps resetting after sending only one set of data, so the intervals become around 15 seconds (this is what it takes the device to reset and reconnect).

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7636
load:0x40078000,len:13904
load:0x40080400,len:5296
entry 0x40080710
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 12:32:35␛[0m
␛[0;32mI (29) boot: chip revision: 3␛[0m
␛[0;32mI (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0␛[0m
␛[0;32mI (40) qio_mode: Enabling default flash chip QIO␛[0m
␛[0;32mI (45) boot.esp32: SPI Speed      : 80MHz␛[0m
␛[0;32mI (50) boot.esp32: SPI Mode       : QIO␛[0m
␛[0;32mI (54) boot.esp32: SPI Flash Size : 16MB␛[0m
␛[0;32mI (59) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (64) boot: Partition Table:␛[0m
␛[0;32mI (68) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (90) boot:  2 factory          factory app      00 00 00010000 003f7a00␛[0m
␛[0;32mI (97) boot:  3 ota_0            OTA app          00 10 00410000 00010000␛[0m
␛[0;32mI (105) boot:  4 ota_1            OTA app          00 11 00420000 00640000␛[0m
␛[0;32mI (113) boot:  5 spiffs           Unknown data     01 82 00a60000 004c4c00␛[0m
␛[0;32mI (120) boot: End of partition table␛[0m
␛[0;32mI (124) boot_comm: chip revision: 3, min. application chip revision: 0␛[0m
␛[0;32mI (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24fb8 (151480) map␛[0m
␛[0;32mI (185) esp_image: segment 1: paddr=0x00034fe0 vaddr=0x3ffbdb60 size=0x03ee0 ( 16096) load␛[0m
␛[0;32mI (191) esp_image: segment 2: paddr=0x00038ec8 vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (192) esp_image: segment 3: paddr=0x000392d4 vaddr=0x40080404 size=0x06d44 ( 27972) load␛[0m
␛[0;32mI (211) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xc594c (809292) map␛[0m
␛[0;32mI (449) esp_image: segment 5: paddr=0x00105974 vaddr=0x40087148 size=0x0c488 ( 50312) load␛[0m
␛[0;32mI (478) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (478) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (479) psram: This chip is ESP32-D0WD␛[0m
␛[0;32mI (483) spiram: Found 64MBit SPI RAM device␛[0m
␛[0;32mI (488) spiram: SPI RAM mode: flash 80m sram 80m␛[0m
␛[0;32mI (493) spiram: PSRAM initialized, cache is in low/high (2-core) mode.␛[0m
␛[0;32mI (500) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (504) cpu_start: Application information:␛[0m
␛[0;32mI (509) cpu_start: Project name:     AWS_IoT_EduKit-Smart-Thermostat␛[0m
␛[0;32mI (516) cpu_start: App version:      b281edf␛[0m
␛[0;32mI (521) cpu_start: Compile time:     Apr  7 2021 12:27:33␛[0m
␛[0;32mI (527) cpu_start: ELF file SHA256:  6fe07958e952a94a...␛[0m
␛[0;32mI (533) cpu_start: ESP-IDF:          3.40200.210118␛[0m
␛[0;32mI (538) cpu_start: Starting app cpu, entry point is 0x40082c30␛[0m
␛[0;32mI (530) cpu_start: App cpu up.␛[0m
␛[0;32mI (1042) spiram: SPI SRAM memory test OK␛[0m
␛[0;32mI (1042) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (1043) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM␛[0m
␛[0;32mI (1049) heap_init: At 3FFCD920 len 000126E0 (73 KiB): DRAM␛[0m
␛[0;32mI (1055) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (1061) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (1068) heap_init: At 400935D0 len 0000CA30 (50 KiB): IRAM␛[0m
␛[0;32mI (1074) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (1079) spiram: Adding pool of 4082K of external SPI memory to heap allocator␛[0m
[... 83 lines cut ...]
␛[0;32mI (19148) Smart_Spaces: setting side LEDs to blue␛[0m
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f09d5  PS      : 0x00060d30  A0      : 0x800d7634  A1      : 0x3ffe2760
A2      : 0x3ffc3cc8  A3      : 0x00000000  A4      : 0x0000ff00  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x800f09d5  A9      : 0x3ffe2740  
A10     : 0x00000000  A11     : 0x0000ff00  A12     : 0x3ffe2760  A13     : 0x0000000c
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

Backtrace:0x400f09d2:0x3ffe2760 0x400d7631:0x3ffe2780 0x400d2510:0x3ffe27a0 0x400d4ef6:0x3ffe27c0 0x400d6615:0x3ffe2800 0x400d685e:0x3ffe2830 0x400d69ca:0x3ffe2890 0x400d72e8:0x3ffe28b0 0x400d73ec:0x3ffe28e0 0x400d45df:0x3ffe2900 0x400d2792:0x3ffe2920 0x4008bcb5:0x3ffe36b0

ELF file SHA256: 6fe07958e952a94a

Rebooting...
ets Jul 29 2019 12:21:46
```

Also, the device lights are not being triggered when the states are changed to COOLING or HEATING (they remain off).

I hope this information is useful.

Cheers,
David

rashedtalukder, 2021-04-08:
Hi @David-Mesa, I'll be investigating the issue today. I haven't seen this regression in my prior tests, but your detailed explanation and the log is very helpful.

rashedtalukder, 2021-04-08:
Issue should be fixed. Please run the commands to get and merge the changes into your local repository:
```
git fetch
git pull
```

Sorry for the challenge you experienced, but your time to report this was very appreciated. I'll leave the issue open until you try it and have it working.

David-Mesa, 2021-04-08:
I'm getting this error:
```
PS C:\Core2-for-AWS-IoT-EduKit\Smart-Thermostat> git pull
error: Your local changes to the following files would be overwritten by merge:
        Smart-Thermostat/main/main.c
Please commit your changes or stash them before you merge.
Aborting
Updating b281edf..3586294
```
Is the main.c file already updated in the code repository? I could replace that file directly if it is the only thing missing. Right?

David-Mesa, 2021-04-08:
Also, do I have to erase the device memory and compile it again? Or can I just compile on topof it once I have the updated code?

rashedtalukder, 2021-04-08:
> I'm getting this error:
>
> ```
> PS C:\Core2-for-AWS-IoT-EduKit\Smart-Thermostat> git pull
> error: Your local changes to the following files would be overwritten by merge: [...]

Yes, you can replace the contents of the file directly. You can also stash your changes if you want by doing a `git stash` and then `git pull`. You can get back to your changes later with `git pop`.

rashedtalukder, 2021-04-08:
> Also, do I have to erase the device memory and compile it again? Or can I just compile on topof it once I have the updated code?

You don't need to erase the memory and go through the build process again. What will happen is that the linker will see the change to the few files and compile a new application binary. We recommend erasing the memory mainly to save yourself from having unwanted cloud service fees with the device continually running and to avoid strange/hard to diagnose behavior from having data from a different application still existing in the memory map.

David-Mesa, 2021-04-08:
Thank you for your swift response, Rashed. It is all working as expected with no rebooting. Lights are working as well with the temperature changes.

closed 2021-04-08 by rashedtalukder