m5stack/Core2-for-AWS-IoT-Kit #52: mbedtls_net_connect returned -0x52

littlehi, 2021-04-12:
Hi,

When we follow this workshop https://edukit.workshop.aws/en/intro-to-alexa-for-iot/building-and-testing-afi.html, to built AFI Demo , it will return "mbedtls_net_connect returned -0x52 "

[0;31mE (698802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(698802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
xf␛[0;31mE (706802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(706802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (714802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(714802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (722802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(722802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (730802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(730802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;33mW (732822) LSSDP: remove timeout SSDP neighbor:  (http://192.168.31.115:39541/upnp/dev/25416%2FA9Y208119/desc) (58932ms)␛[0m
␛[0;31mE (738802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(738802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (746802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(746802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (754802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(754802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (762802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(762802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com
␛[0;31mE (770802) aws_iot: failed! mbedtls_net_connect returned -0x52␛[0m
:8883␛[0m(770802) [http_transport]: Error(-23) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com

rashedtalukder, 2021-04-12:
Hi @littlehi, I'll take a look at it, but this looks like a support question. Please post support questions in the AWS IoT EduKit tutorial GitHub discussions: https://github.com/aws-samples/aws-iot-edukit-tutorials/discussions

rashedtalukder, 2021-04-12:
Customer moved to support discussion: https://github.com/aws-samples/aws-iot-edukit-tutorials/discussions/26

closed 2021-04-12 by rashedtalukder