m5stack/Core2-for-AWS-IoT-Kit #55: Setting motor strength to 0 when it's off will turn it on

nuchi, 2021-05-05:
**To reproduce:** call `Core2ForAWS_Motor_SetStrength(0)` when the motor is off.
**What happens:** The motor is turned on, with voltage at the lowest level. Note that the `else if` branch below is taken because the `strength` parameter is `0` and `motor_state` is `false`.
**What I expect to happen:** that this is a no-op and the motor remains off.

https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/ce8e83ad4222e510937b0cd9934f0d244aa25203/Blinky-Hello-World/components/core2forAWS/core2forAWS.c#L85-L99

rashedtalukder, 2021-05-06:
Hi Nuchi, thank you for reporting I see the issue and will patch that in the next few days.

rashedtalukder, 2021-05-28:
Fixed: https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/32652af72aa57dc79240938626c0a66f61ca48b1#diff-4f468c334508bc73f5ff8c8236a28fbd315aa0e57ae00101289881b17cdc9a62

closed 2021-05-28 by rashedtalukder