m5stack/Core2-for-AWS-IoT-Kit #56: Arrow Key Inputs Not working for MenuConfig

RCNewton85, 2021-05-07:
https://edukit.workshop.aws/en/blinky-hello-world/connecting-to-aws.html

Windows 10 - Visual Studio - PlatformIO Terminal 

Here you will set the configuration. Use the direction keys on your keyboard to go to Component config –> Amazon Web Services IoT Platform 

Arrow Keys do nothing. Other Keys work.

RCNewton85, 2021-05-07:
Tried on Screen Keyboard as well. Unresponsive as well.

rashedtalukder, 2021-05-07:
I found this issue: https://github.com/platformio/platform-espressif32/issues/423

Do you have a US keyboard with the standard layout? Is there a VSCode configuration for the keyboard that might have been changed? Seems like there might be a different mapping for your keys. According to that issue, **J** & **K** buttons are mapped to up and down for those users. Does that seem to work?