m5stack/Core2-for-AWS-IoT-Kit #58: Doc needs to be clear on QR Code

erbenmo, 2021-05-16:
I was following https://edukit.workshop.aws/en/getting-started/run-rainmaker.html

I am not seeing a QR Code initially. I had to click the refresh button for the QR code to show on my computer. The documentation is not clear on that.

qaoo8, 2021-05-17:
The same problem here, everything runs with no errors, the monitor shows the temperature repeatedly, but no QR code shows on my computer.

rashedtalukder, 2021-05-17:
Hi @erbenmo and @qaoo8, is it actually showing up, but you need to scroll up in the terminal window?

its-mirus-lu, 2021-05-25:
@erbenmo @qaoo8 I ran into the same issue. I solved it by following these steps:

1) Run `pio run --environment core2foraws --target upload`
2) Run `pio run --environment core2foraws --target monitor`
3) Select `/dev/cu.SLAB_USBtoUART`
4) Click the reset button at the bottom of the CORE2 device

The terminal will update and eventually show the QR code.

rashedtalukder, 2021-05-25:
@its-mirus-lu, do you happen to have the terminal output for when it wasn't displaying the QR code? I've never seen this issue and am not able to reproduce. Uploading the firmware should cause the device to reboot automatically once complete.

Also, which OS are all of you using?

qaoo8, 2021-05-26:
@rashedtalukder, as I mention  **" the monitor shows the temperature repeatedly"** the monitor keeps scrolling down while updating the temperature from the device; The CLI terminal is kept reflash, the QR code is no showing before I clicking the reset button of the device, but it shows up after pushing the reset button.
my os: Ubuntu 20.04.

compusam, 2021-07-16:
> @rashedtalukder, as I mention **" the monitor shows the temperature repeatedly"** the monitor keeps scrolling down while updating the temperature from the device; The CLI terminal is kept reflash, [...]

You have to press the reset button and that's it
The QR code appear in the console.

rashedtalukder, 2021-07-16:
This commit fixes that issue: https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/0b6903301d5b7020dd29c66aa3d70771be22088b

closed 2021-07-16 by rashedtalukder