m5stack/Core2-for-AWS-IoT-Kit #60: ESP Alexa iOS app crashing on iPhone11 ios14.6

akiyoshi-cpu, 2021-06-09:
I installed ALEXA App on M5stack core2 for AWS. It seems work correctly. I can see screen brightness slider on screen and red light is flashing on the side.
ESP Alexa App is up and running on iPhone 11, ios14.6.
I press button "Add New Device" on iPhone
4)My M5 is listed in devices via Bluetooth as "ESP-Alexa-7700"
Selected device in list
Amazon Alexa screen appear with "Proof of possession" and "abcd1234" in text field, press button "done"
ESP Alexa app disappear from iPhone screen...
Any idea what to do?

rashedtalukder, 2021-06-12:
Hi @akiyoshi-cpu,

Have you tried to completely delete the phone app (and the app data) and then reinstall it? There might be something cached in the app. Another possibility is that the app does not have all the permissions it needs.

rashedtalukder, 2021-06-29:
Closing since there hasn't been a response. Please reopen if you're still having this issue and have additional information you can provide.

akiyoshi-cpu, 2021-06-30:
Dear Sir, Thank you very much! It worked! I tried to delete it and install the application again as you said. Then it worked.

RobertoD91, 2021-07-17:
I have the same issue
The app "ESP Alexa" crash after entering PoP text.
iOS 14.6 - iPhone 6s
reinstalling has no effect

rashedtalukder, 2021-07-17:
Would you mind trying the following
1. Uninstall the app on your phone
2. Reboot your phone
3. Wipe the certificates and firmware on your device with: pio run -e core2foraws -t erase
4. Reupload the certificates
5. Reupload the firmware
6. Reinstall the app on your phone

RobertoD91, 2021-07-17:
Thanks, now it works.

Additional notes:
-On an ipad air 2 the app never crashed.
-After following your instructions the app also worked on the iphone 6s. However it did not detect the device in the "manage devices" section.
-After repeating your instructions again the device works correctly.

closed 2021-07-17 by rashedtalukder