m5stack/Core2-for-AWS-IoT-Kit #61: Original Getting-Started project get grabRef.cmake error without any change

muriloplinio, 2021-06-11:
**Following this tutorial from the first step:**
_https://edukit.workshop.aws/en/getting-started.html_

**After executing this command:**
_pio run --environment core2foraws_

**I had this error:**
_CMake Error at /Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):_

I am using MacOS Big Sur ver 11.2.3 in a M1 Mac.
I am using Visual Studio Code Version: 1.57.0 for Apple Silicon (M1) model.

**Makefile:**

_PROJECT_NAME := AWS_IoT_EduKit-Getting_Started
PROJECT_VER := 1.0.1
include $(IDF_PATH)/make/project.mk_

**CMakeList.txt:**
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(AWS_IoT_EduKit-Getting_Started VERSION 1.0.3)

**************************************************************************
 

**This is the complete terminal screen:**

MacBook-Air:Getting-Started muriloplinio$ pio run --environment core2foraws
Processing core2foraws (platform: espressif32@3.2.0; framework: espidf; board: m5stack-core2)
-----------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-core2.html
PLATFORM: Espressif 32 (3.2.0) > M5Stack Core2
HARDWARE: ESP32 240MHz, 6.25MB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40200.210118 (4.2.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py log main micro-ecc partition_table soc spi_flash xtensa
-- Component paths: /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader_support /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/efuse /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp32 /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp_common /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esp_rom /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/esptool_py /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/log /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader/subproject/main /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/bootloader/subproject/components/micro-ecc /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/partition_table /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/soc /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/spi_flash /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/components/xtensa
-- Configuring incomplete, errors occurred!
See also "/Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/CMakeOutput.log".

CMake Error at /Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/build.cmake:68 (git_describe)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/idf.cmake:49 (__build_init)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:20 (include)

CMake Error at /Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /Users/muriloplinio/m5stack/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/project.cmake:48 (git_describe)
  /Users/muriloplinio/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/project.cmake:391 (__project_get_revision)
  CMakeLists.txt:27 (project)

====================================================== [FAILED] Took 5.47 seconds ======================================================_

muriloplinio, 2021-06-11:
I copied the file "head_ref" 
from 
.../Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/CMakeFiles/git-data
to 
.../Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/bootloader/CMakeFiles/git-data

Now it is working!

rashedtalukder, 2021-06-12:
Hi @muriloplinio, 
Did you clone or download the project? If you cloned the repo, this issue shouldn't come up. I'll dive further after receiving your response.

muriloplinio, 2021-06-14:
Hi, @rashedtalukder! Thank you!
I followed the tutorial and I used the command:

"Paste https://github.com/m5stack/Core2-for-AWS-IoT-EduKit.git into the text field and then select the location you want to save the project in."

so I clone!

closed 2021-07-16 by rashedtalukder