m5stack/Core2-for-AWS-IoT-Kit #62: Intro To Alexa for IoT - Build fails

haybnann, 2021-06-18:


rashedtalukder, 2021-06-21:
Hi @haybnann , is it still failing? Do you have any logs or additional information to provide?

maulsparks, 2021-06-26:
M5tack was configured and after about  3 weeks of inactivity, the device screen is blank and I can hear a clicking noise. Efforts to power on/power off and reset have no effect. Just a blank screen and a clicking noise. Any ideas on how to fix?

rashedtalukder, 2021-06-26:
@maulsparks if you could in the future post support questions and not possible bugs in its own support discussion thread on https://github.com/aws-samples/aws-iot-edukit-tutorials/discussions instead of posting it as a reply in a issue that isn’t related, it helps keep things organized and easier for others in the future to find the right help. 

That said, there’s nothing wrong with the device. You probably just don’t have any application loaded in your device after erasing as noted in here:  https://edukit.workshop.aws/en/getting-started/run-rainmaker.html#erasing-the-firmware-with-platformio . So the device just reboots over and over causing the speaker amplifier to power cycle on and off repeatedly. To have it stop doing that, you just need to upload one of the provided projects like in here: https://edukit.workshop.aws/en/getting-started/run-rainmaker.html

closed 2021-07-16 by rashedtalukder