m5stack/Core2-for-AWS-IoT-Kit #63: Factory-Firmware build, failed with a bricked device

xiaolaba, 2021-06-21:
hi, it is new to play with this Core2-for-AWS-IoT-EduKit. Something happened after upload, the device looks bricked, no screen, no boot. Plugging the USB is able to see COM5 appeared, Direct uses esptool.exe to read_mac and flash_id are both ok with response from the serial port, and then trying to restore factory firmware, it failed on build as well, any help or a malfunction unit ?

log is following,

`Windows PowerShell
Copyright (C) Microsoft Corporation. 著作權所有，並保留一切權利。

請嘗試新的跨平台 PowerShell https://aka.ms/pscore6

PS C:\Users\user0\M5stackAWS\Core2-for-AWS-IoT-EduKit\Getting-Started> cd ..
PS C:\Users\user0\M5stackAWS\Core2-for-AWS-IoT-EduKit> cd Factory-Firmware
PS C:\Users\user0\M5stackAWS\Core2-for-AWS-IoT-EduKit\Factory-Firmware> pio run -e core2foraws -t flash -t upload
Processing core2foraws (platform: espressif32@3.2.1; framework: espidf; board: m5stack-core2)      
---------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/framework-espidf @ ~3.40201.0
Tool Manager: Error: Please read http://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.nm1.platformio.org', port=443): Max retries exceeded with url: /tools/17/3c/7c52517621567d1b9a70aeeef8db0486e2abe9ce9b865c3695cc38e34c1f/framework-espidf-3.40201.210513.tar.gz (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x045DEC10>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read http://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.ns1.platformio.org', port=443): Max retries exceeded with url: /tools/17/3c/7c52517621567d1b9a70aeeef8db0486e2abe9ce9b865c3695cc38e34c1f/framework-espidf-3.40201.210513.tar.gz (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x045DE970>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))
Tool Manager: Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-espidf @ 3.40201.210513 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-core2.html
PLATFORM: Espressif 32 (3.2.1) > M5Stack Core2
HARDWARE: ESP32 240MHz, 6.25MB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, 
olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40201.210513 (4.2.1)
 - tool-cmake 3.16.5
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
*** Do not know how to make File target `flash' (C:\Users\user0\M5stackAWS\Core2-for-AWS-IoT-EduKit\Factory-Firmware\flash).  Stop.
================================== [FAILED] Took 545.41 seconds ==================================
PS C:\Users\user0\M5stackAWS\Core2-for-AWS-IoT-EduKit\Factory-Firmware> `

rashedtalukder, 2021-06-21:
Hi @xiaolaba,

As you can see from the last line in the error:
```
Do not know how to make File target flash'
````

Meaning the target you are giving it with `-t flash` is not valid and you should remove that part of the command. If you are just trying to upload (to flash the device), just give the command `pio run -e core2foraws -t upload`.

rashedtalukder, 2021-06-21:
And also, your device is not bricked.

xiaolaba, 2021-06-22:
thanks. we believed the AWS doc has correction done. it works now.

https://edukit.workshop.aws/en/getting-started/run-rainmaker.html

> 
> 
> And also, your device is not bricked.

thanks. we believed the AWS doc has correction done. it works now.

https://edukit.workshop.aws/en/getting-started/run-rainmaker.html

closed 2021-06-22 by xiaolaba