m5stack/Core2-for-AWS-IoT-Kit #70: Digital Port A cannot be used on AWS IoT EduKit

Raunak-Singh-Inventor, 2021-07-20:
I am trying to connect a external grove sensor (https://wiki.seeedstudio.com/Grove-Finger-clip_Heart_Rate_Sensor_with_shell/) to the Port A (red port) on the AWS IoT EduKit. 

I couldn't find a preprocessor macro for Port A  (there are preprocesor macros for Port B and Port C, but no preprocesor macro for Port A), so I initially tried getting the data from the sensor  from GPIO_NUM_32/GPIO_NUM_33 through the `Core2ForAWS_Port_Read` func in the Core2 for AWS BSP (https://edukit.workshop.aws/en/api-reference/expports.html) through this code:
```
void getHeartrateInput(void *parameter) {
    printf("initialized getHeartrateInput task\n"); 
    while(true) {
        Core2ForAWS_Port_PinMode(GPIO_NUM_33, INPUT); // set pinmode to i/o pin connected to red i2c port
        printf("(getHeartrateInput) Heartrate: %d\n",Core2ForAWS_Port_Read(GPIO_NUM_33)); 
        vTaskDelay(pdMS_TO_TICKS(1000)); // to make the output in the monitor more readable
    }
}
```
But, it didn't work.

Next, I tried this code, which uses the i2c library in the core2foraws, but this also doesn't work:
```
void getHeartrateInput(void *parameter) {
    printf("initialized getHeartrateInput task");
    while(true) {
        uint8_t heart_sensor_address = 0xA0 >> 1; 

        I2CDevice_t heart_sensor_device = i2c_malloc_device(I2C_NUM_1, GPIO_NUM_32, GPIO_NUM_33, 100000, heart_sensor_address);

        uint8_t *data = NULL;
        uint8_t register_addr = 0x01; 
        esp_err_t err = i2c_read_byte(heart_sensor_device, register_addr, data);

        if(err == ESP_OK){
            printf("Heartrate: %hhn",data);
        }
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```
Is it possible to connect an external grove sensor to Port A? Please help.

Raunak-Singh-Inventor, 2021-07-21:
Here is the registers page in the datasheet for the heartrate sensor:
[image]
I2C is labeled as `0x4000 5800`. How should I use it in my C program?

rashedtalukder, 2021-08-13:
Fixed in https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/332ac2bd5afc3837327af973733deaee10491523 and added APIs to read/write to I2C device.

closed 2021-08-13 by rashedtalukder