m5stack/Core2-for-AWS-IoT-Kit #73: There are two speaker.h files.(Alexa_for_IoT_Intro)

mongonta0716, 2021-08-24:
It may be due to the way my application is modified, but the presence of two speaker.h in the component may cause a build error during application development.

I worked around this by renaming the core2forAWS/speaker file to a different name.(core2speaker.h, core2speaker.c)

[components/core2forAWS/speaker](https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/tree/master/Alexa_for_IoT-Intro/components/core2forAWS/speaker)

[components/voice_assistant/include](https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/tree/master/Alexa_for_IoT-Intro/components/voice_assistant/include)

closed 2021-08-24 by rashedtalukder

mongonta0716, 2021-08-24:
360d17f

Isn't there a link error when you modify the voice_assistant one?

rashedtalukder, 2021-08-24:
You'll see in the commit that the referenced includes in the **app_main.c** and **va_button.c** files have been updated. The CMakeLists.txt files include all files in the includes/ directory for link time, so there's no need to modify thoughts. The CMake build system will automatically include the updated files during link-time.

mongonta0716, 2021-08-24:
I understand.

In my case, I couldn't link with libaia.a when linking and the build didn't proceed. Maybe my app was affected by changing app_main.c to app_main.cpp.

Great project. I am enjoying using it. Thank you.
(I forgot my first greeting. Sorry.

rashedtalukder, 2021-08-24:
Oh ok I see. You'll most likely to make a few modifications to work with a cpp linker/compiler. I won't be as straight forward as renaming the files.

Glad you're liking it so far. Looking forward to seeing what you build with it 😄