m5stack/Core2-for-AWS-IoT-Kit #75: Error building RainMaker Agent firmware

Pedotech, 2021-09-30:
Hi
When I run      
pio run --environment core2foraws 
 
I get error messages:

Generating assembly for certificate bundle...
CMake Error at /Users/pedrocorreia/.platformio/packages/framework-espidf@3.40200.210118/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
  file failed to open for reading (No such file or directory):

    /Users/pedrocorreia/aws_set/Core2-for-AWS-IoT-EduKit/Getting-Started/.pio/build/core2foraws/x509_crt_bundle

*** Error 1
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
*** [.pio/build/core2foraws/.pio/build/core2foraws/x509_crt_bundle.o] Source `.pio/build/core2foraws/x509_crt_bundle.S' not found, needed by target `.pio/build/core2foraws/.pio/build/core2foraws/x509_crt_bundle.o'.
================================================================================================ [FAILED] Took 11.01 seconds =

My vsc and pio instalations are otherwise OK. 
I tried on Linux and get the same error.

Is there a missing file in gitclone?

I am stuck and very much in need of help

Pedro

rashedtalukder, 2021-10-01:
@Pedotech @jaimeskatz, both of you appear to be having the same issue and both are on a UNIX system. I've been able to replicate this on Ubuntu, macOS, and Windows 10. This appears to be a regression in the latest version of PlatformIO. I am going to continue to investigate this and see if I can find the source of the CMake error.

rashedtalukder, 2021-10-01:
This appears to be a certificate expiry error https://github.com/espressif/esp-idf/issues/7621 in [this root CA](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cacrt_all.pem) that mbedTLS uses.

I'll be applying a temporary workaround commit to this repo shortly.

rashedtalukder, 2021-10-01:
@Pedotech @jaimeskatz, please either reclone this repo or use:

```bash
git fetch
git pull
```

If you receive any errors for your files needing to be changed, if you're ok with wiping all the changes out, you can do a `git reset --hard origin/master` and then executing `git pull`.

I'm going to leave this open until the fix has been applied to the ESP-IDF with updated certs.

Pedotech, 2021-10-01:
Hi Rashedtalukder
Problem solved! 
Thank you
Pedro

closed 2021-10-01 by Pedotech

jaimeskatz, 2021-10-01:
Worked for me also. Thank you!