m5stack/Core2-for-AWS-IoT-Kit #82: All examples panic after flashing

kirkrudolph, 2021-11-08:
Hello,

I followed the demos on my IoT-EduKit using ProjectIO a while ago without any issues. Today I did a fresh clone of this repo but my unit continuously panics after flashing any example with `idf.py flash`. I noticed the README says IDF v4.2 while I have IDF v4.3.1 installed. 

Here's a debug level log of the console output via the `idf.py monitor` command:
[log.M5Core2ForAWSDemo.20211107185252.txt](https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/files/7493556/log.M5Core2ForAWSDemo.20211107185252.txt)

Any help would be appreciated. Let me know if a verbose log would help.

Thanks,
Kirk

kirkrudolph, 2021-11-08:
Update: I was able to succussfully build and flash the examples in this repo without re-installing IDF v4.2 via:

    docker run --rm --device /dev/ttyUSB0 -v $PWD:/project -w /project -it espressif/idf:v4.2 idf.py flash

I'm curious what component(s) in these examples only work with IDF v4.2? Are there any plans to make these examples work with future IDF versions or will they always be pinned to IDF v4.2?

rashedtalukder, 2021-11-08:
Hi @kirkrudolph,
The issue you're seeing comes from our SPI driver and the mutex in it. We are aware of the issue and looking into a long-term fix to the entire BSP. There are also changes made to the RMT APIs in ESP-IDF v4.3+ that cause other issues as well.

kirkrudolph, 2021-11-13:
Thanks for the update, @rashedtalukder. Everything has been working great with IDF v4.2.

closed 2021-11-13 by kirkrudolph