m5stack/Core2-for-AWS-IoT-Kit #83: "Light" virtual device is not created

omerio, 2021-11-18:
Alexa works fine on the EduKit, however the Light virtual device is not created as described in the workshop and also not recognized by Alexa. Very likely related to issue: https://github.com/espressif/esp-va-sdk/issues/130

rashedtalukder, 2021-11-19:
Hi Omerio, you are right and there is an issue where the friendlyName parameter seems to not not being used. I believe I've identified the issue in the endpointId parameter, however the code for this is in the proprietary binary supplied by Espressif. We've been working with Espressif and the Alexa team on this issue and they are testing some of the proposed changes. Should see updates on their side and ours if that goes well.

Please stay tuned.

rashedtalukder, 2021-11-22:
As a temporary workaround, you can rename the light device to "Light" in the Alexa app and it will work from that point.