m5stack/Core2-for-AWS-IoT-Kit #84: LVGL crash

Leo-Dev, 2021-12-20:
Hi @rashedtalukder ,

We are facing LVGL crash issue as below: the crash is random, sometimes few mins or over an hour.
```
0x401131f9: lv_draw_label at /components/core2forAWS/tft/lvgl/lvgl/src/lv_draw/lv_draw_label.c:121

0x40124ba2: lv_label_design at /components/core2forAWS/tft/lvgl/lvgl/src/lv_widgets/lv_label.c:1347

0x4010f47e: lv_refr_obj at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:694 (discriminator 1)
                                                                                                                                                                                                                                                            0x4010f54b: lv_refr_obj_and_children at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:651

0x4010f854: lv_refr_area_part at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:567

0x4010f9d9: lv_refr_area at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:478 (discriminator 2)

0x4010fac2: lv_refr_areas at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:400

0x4010fb24: _lv_disp_refr_task at /components/core2forAWS/tft/lvgl/lvgl/src/lv_core/lv_refr.c:199

0x4011bf09: lv_task_exec at /components/core2forAWS/tft/lvgl/lvgl/src/lv_misc/lv_task.c:386 (discriminator 1)

0x4011bff4: lv_task_handler at /components/core2forAWS/tft/lvgl/lvgl/src/lv_misc/lv_task.c:134

0x40109da7: guiTask at /components/core2forAWS/core2forAWS.c:383

0x4008c2e1: vPortTaskWrapper at /home/leo/project/m5stack/firmware/esp_idf/esp-idf/components/freertos/xtensa/port.c:143

```
Info:

1. Core2-for-AWS-IoT-EduKit/Factory-Firmware:  commit 40e9c3ac5d14e55d8f1d28d453141174c5978ed1
2. esp-idf: commit c40f2590bf759ff60ef122afa79b4ec04e7633d2 (HEAD, tag: v4.2)

And we are not using "platformio", and use idf.py the bulid and flash image to m5stack.

Thank you.

rashedtalukder, 2021-12-21:
That's fascinating. We're actually [updating the BSP](https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/tree/BSP-dev) from the bottom up currently and we'll refactor all of the firmware after that's done. I am hope that it will fix that issue as a by-product of the latest round of updates since we find a few ways to optimize the display task using libraries provided by the community.

iloop2020, 2021-12-21:
Hi @rashedtalukder 

Thank you your reply.
That is great you will have newer version.

For now, do you have any suggestions for me to workaround the issue ?

Thank you.

rashedtalukder, 2021-12-21:
Which sensor screen is it on when it crashes? Home/any? Or one of them specifically?

closed 2026-10-24 by rashedtalukder