m5stack/Core2-for-AWS-IoT-Kit #85: M1 Support

DarkSector, 2021-12-25:
Microchip Cryptoauthlib package is not buildable using Apple M1. Is there a work around for this? This comes up when trying to register_thing.

cryptoauthlib also only supports python 3.7 nothing beyond that. 

```
pio run -e core2foraws-device_reg -t register_thing > ~/Desktop/output.txt
  ERROR: Command errored out with exit status 1:
   command: /Users/pronoyc/.platformio/penv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py'"'"'; __file__='"'"'/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-wheel-p8uuetld
       cwd: /private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/
  Complete output (302 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.14-arm64-3.8
  creating build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/tng.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/library.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/device.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/iface.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atjwt.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/__init__.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcab.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcacert.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/sha206_api.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/exceptions.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/talib.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcaenum.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/status.py -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  running egg_info
  writing cryptoauthlib.egg-info/PKG-INFO
  writing dependency_links to cryptoauthlib.egg-info/dependency_links.txt
  writing requirements to cryptoauthlib.egg-info/requires.txt
  writing top-level names to cryptoauthlib.egg-info/top_level.txt
  reading manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching 'release_notes.md'
  writing manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
  copying cryptoauthlib/cryptoauth.json -> build/lib.macosx-10.14-arm64-3.8/cryptoauthlib
  running build_ext
  Traceback (most recent call last):
    File "/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py", line 187, in build_extension
      subprocess.check_output(['cmake', '--build', '.'] + build_args, cwd=os.path.abspath(self.build_temp), stderr=subprocess.STDOUT)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 415, in check_output
      return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 516, in run
      raise CalledProcessError(retcode, process.args,
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release']' returned non-zero exit status 2.
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py", line 213, in <module>
      setup(
    File "/Users/pronoyc/.platformio/penv/lib/python3.8/site-packages/setuptools/__init__.py", line 165, in setup
      return distutils.core.setup(**attrs)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/Users/pronoyc/.platformio/penv/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/command/build.py", line 135, in run
[... 548 lines cut ...]
    make[2]: *** [/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/build/lib.macosx-10.14-arm64-3.8/cryptoauthlib/libcryptoauth.dylib] Error 1
    make[1]: *** [CMakeFiles/cryptoauth.dir/all] Error 2
    make: *** [all] Error 2
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/pronoyc/.platformio/penv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py'"'"'; __file__='"'"'/private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-install-uxyw9m8o/cryptoauthlib_dadf2d8705064bcea0d3ad74bd64c482/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/j3/csj45mw17tlb2ggw27_3vfc00000gn/T/pip-record-h98f6cna/install-record.txt --single-version-externally-managed --compile --install-headers /Users/pronoyc/.platformio/penv/include/site/python3.8/cryptoauthlib Check the logs for full command output.
CalledProcessError: Command '['/Users/pronoyc/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-cache-dir', '-r', '/Users/pronoyc/Code/AWSIoT/edukit/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/utilities/AWS_IoT_registration_helper/requirements.txt']' returned non-zero exit status 1.:
  File "/Users/pronoyc/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 189:
    env.SConscript(item, exports="env")
  File "/Users/pronoyc/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/pronoyc/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/pronoyc/Code/AWSIoT/edukit/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/register_thing.py", line 39:
    install_dependencies()
  File "/Users/pronoyc/Code/AWSIoT/edukit/Core2-for-AWS-IoT-EduKit/Blinky-Hello-World/register_thing.py", line 37:
    subprocess.check_call( [ env.get( 'PYTHONEXE' ), '-m', 'pip', 'install', '--no-cache-dir', '-r',  os.path.join( cwd, 'utilities', 'AWS_IoT_registration_helper', 'requirements.txt') ] )
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 364:
    raise CalledProcessError(retcode, cmd)
========================== [FAILED] Took 8.98 seconds ==========================
Environment             Status    Duration
----------------------  --------  ------------
core2foraws-device_reg  FAILED    00:00:08.984
==================== 1 failed, 0 succeeded in 00:00:08.984 ====================
```

DarkSector, 2021-12-26:
Doesn't build with python 3.7.12 on M1 with rosetta either

rashedtalukder, 2021-12-27:
Hi @DarkSector, I think I found a solution to this since I just received my new M1 macbook. You'll need to use the Intel x86 version of VSCode for PlatformIO to work correctly with Rosetta: https://code.visualstudio.com/sha/download?build=stable&os=darwin

After that, I was able to use the command `pio run -e core2foraws-device_reg -t register_thing` and it worked perfectly. Do note that we update the repository recently to add the PIO extra script with the register_thing target. Make sure to pull the latest code.

DarkSector, 2021-12-27:
Great idea! Let me try that

DarkSector, 2021-12-27:
Okay that worked. For posterity, here are the steps to get it working with Apple M1 silicon

1. Install the x86 version of vscode
2. Install the x86 version of homebrew (you'll need it for packages like openssl, cmake, ffi etc.). 
3. If you're getting errors with arm64 dylibs - uninstall pyopenssl and cryptography from the platformIO environment. 
4. If you use AWS SSO add access keys as environment variables.

Should be able to build it then.

closed 2021-12-27 by DarkSector

rashedtalukder, 2021-12-27:
Thank you for those additional details. We now have ARM M1 AWS EC2 instances to test against so we can document the steps properly for M1 and ensure it works for all users.

Overall, I'm glad that the workaround for now is relatively straightforward. Hopefully we see native ARM M1 support for all dependencies instead of using Rosetta.

DarkSector, 2021-12-27:
Is there a possibility of a docker image with all the dependencies baked in? It would be great to just pass the serial port to the container and never have to worry about installation etc.

rashedtalukder, 2021-12-27:
Yes, this is something we have been wracking our brains trying to solve and have made multiple trials. However this can cause new complications on Windows due to Docker on Hyper-V.