m5stack/Core2-for-AWS-IoT-Kit #86: Does not work on M1 Mac

PeterJBurke, 2022-01-09:
pip install cryptoauthlib fails on M1 mac.
See also https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/issues/85
https://github.com/MicrochipTech/cryptoauthlib/issues/281

 Here is the error message I get:
peterburke@Peters-MacBook-Air Blinky-Hello-World %     pip install cryptoauthlib
Collecting cryptoauthlib
  Using cached cryptoauthlib-20211006.tar.gz (438 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: cryptoauthlib
  Building wheel for cryptoauthlib (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/peterburke/.platformio/penv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/setup.py'"'"'; __file__='"'"'/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-wheel-avzy9plw
       cwd: /private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/
  Complete output (302 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.14.6-arm64-3.8
  creating build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/tng.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/library.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/device.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/iface.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atjwt.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/__init__.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcab.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcacert.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/sha206_api.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/exceptions.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/talib.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/atcaenum.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  copying cryptoauthlib/status.py -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  running egg_info
  writing cryptoauthlib.egg-info/PKG-INFO
  writing dependency_links to cryptoauthlib.egg-info/dependency_links.txt
  writing requirements to cryptoauthlib.egg-info/requires.txt
  writing top-level names to cryptoauthlib.egg-info/top_level.txt
  reading manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching 'release_notes.md'
  writing manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
  copying cryptoauthlib/cryptoauth.json -> build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib
  running build_ext
  Traceback (most recent call last):
    File "/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/setup.py", line 187, in build_extension
      subprocess.check_output(['cmake', '--build', '.'] + build_args, cwd=os.path.abspath(self.build_temp), stderr=subprocess.STDOUT)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 415, in check_output
      return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 516, in run
      raise CalledProcessError(retcode, process.args,
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release']' returned non-zero exit status 2.
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/setup.py", line 213, in <module>
      setup(
    File "/Users/peterburke/.platformio/penv/lib/python3.8/site-packages/setuptools/__init__.py", line 165, in setup
      return distutils.core.setup(**attrs)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/Users/peterburke/.platformio/penv/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
[... 539 lines cut ...]
    [ 98%] Building C object CMakeFiles/cryptoauth.dir/hal/hal_linux_uart_userspace.c.o
    [100%] Linking C shared library /private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/build/lib.macosx-10.14.6-arm64-3.8/cryptoauthlib/libcryptoauth.dylib
    ld: warning: pointer not aligned at address 0x2836B (_g_tflxtls_cert_def_4_device + 99 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tflxtls_cert_def_4_device.c.o)
    ld: warning: pointer not aligned at address 0x28361 (_g_tflxtls_cert_def_4_device + 89 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tflxtls_cert_def_4_device.c.o)
    ld: warning: pointer not aligned at address 0x283C3 (_g_tnglora_cert_def_1_signer + 80 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_1_signer.c.o)
    ld: warning: pointer not aligned at address 0x28441 (_g_tnglora_cert_def_2_device + 99 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_2_device.c.o)
    ld: warning: pointer not aligned at address 0x28437 (_g_tnglora_cert_def_2_device + 89 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_2_device.c.o)
    ld: warning: pointer not aligned at address 0x2842E (_g_tnglora_cert_def_2_device + 80 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_2_device.c.o)
    ld: warning: pointer not aligned at address 0x284A2 (_g_tnglora_cert_def_4_device + 89 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_4_device.c.o)
    ld: warning: pointer not aligned at address 0x28499 (_g_tnglora_cert_def_4_device + 80 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tnglora_cert_def_4_device.c.o)
    ld: warning: pointer not aligned at address 0x2850D (_g_tngtls_cert_def_1_signer + 89 from CMakeFiles/cryptoauth.dir/private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-install-pv9wthdb/cryptoauthlib_8b5200c2d9b54c6193a75ea3867e9438/app/tng/tngtls_cert_def_1_signer.c.o)
[... cut ...] ('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/xg/_zygyhl957x91rl34w_6nw5c0000gn/T/pip-record-agok2izz/install-record.txt --single-version-externally-managed --compile --install-headers /Users/peterburke/.platformio/penv/include/site/python3.8/cryptoauthlib Check the logs for full command output.
peterburke@Peters-MacBook-Air Blinky-Hello-World %

rashedtalukder, 2022-01-10:
Take a look at:
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/issues/85#issuecomment-1001280007

Downloading and using the x86 version of PlatformIO should let you move forward.

rashedtalukder, 2022-03-05:
I know this took a while to resolve, but https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/commit/2b4287b16b506d217d98df001efc4c48479efffc should fix the issue you were facing on an M1 Mac. This has been pretty challenging to be able to systematically test since we normally use VMs and this is currently not possible.

closed 2022-03-08 by rashedtalukder