m5stack/Core2-for-AWS-IoT-Kit #87: Keeps on rebooting with cryptoauthlib related error

gotcarriedaway, 2022-02-06:
When I go through AWS IoT EduKit tutorial, after one project , for example getting-started , is built and uploaded to the m5stack, I start to monitor the serial output and the below error is shown in the terminal viewpoint:

components/esp-cryptoauthlib/cryptoauthlib/lib/calib/calib_read.c:103:f0:calib_read_zone - execution failed
components/esp-cryptoauthlib/cryptoauthlib/lib/calib/calib_read.c:619:f0:calib_read_zone - falied
␛[0;32mI (22645) atecc608: Initializing ATECC608 secure element␛[0m
␛[0;32mI (22655) atecc608: *FAILED* atcab_init returned f0␛[0m
␛[0;31mE (22655) Core2forAWS: ATECC608 secure element initialization error!␛[0m

abort() was called at PC 0x400d5327 on core 0

Then the m5stack starts rebooting and every 20 seconds, the whole above process repeats.

and the screen on m5stack is always blank ,just blink for one second when rebooted.

So the tutorial could not be carried on. Please help to take a look at this issue. Thanks.

closed 2022-02-07 by gotcarriedaway

li3p, 2022-05-09:
I encountered the same situation, would  you help me to find how to resolve this if you found the cause.

gotcarriedaway, 2022-05-09:
> I encountered the same situation, would  you help me to find how to resolve this if you found the cause.

It turned out to be not rightly plugged in between the screen and the base part. You can try open it and make the pin insert to the right slot.