m5stack/Core2-for-AWS-IoT-Kit #91: ATECC608 config?

torntrousers, 2022-04-28:
Hi, is there any info somewhere about what is the configuration of the ATECC608 secure element thats in the EduKit - e.g. what slots are available for use for keys and writable to generate new keys or for storing data like certificates?

Sorry if I've missed this and its in doc page somwhere , I have looked about but can't see anything.

Thanks!

rashedtalukder, 2022-04-29:
Hi @torntrousers, we have a un-modified ATECC608B Trust&GO. You can view the datasheet here: https://ww1.microchip.com/downloads/en/DeviceDoc/ATECC608B-CryptoAuthentication-Device-Summary-Data-Sheet-DS40002239B.pdf

You can use the [CryptoAuthLib](https://github.com/microchiptech/cryptoauthlib) library to perform operations to the chip itself. The library and ported HAL can be found in the **components/esp-cryptoauthlib** folder. The headers for cryptoauthlib are included in your projects from the **core2foraws.h** file and if the secure element is enabled in the menu (it is by default).

torntrousers, 2022-05-05:
I found what i needed here: https://onlinedocs.microchip.com/pr/GUID-59F7F830-64CA-4EE1-A21D-C5745DEC9509-en-US-1/index.html?GUID-5118176F-10E9-41D3-9FDD-E36F38172F9F:

[image]

closed 2022-05-05 by torntrousers