m5stack/Core2-for-AWS-IoT-Kit #93: ERROR: Unable to verify the server's certificate.

Raunak-Singh-Inventor, 2022-05-15:
Trying to connect to AWS IoT Core by using the code from `Blinky-Hello-World` project in my project. Here is the error I am getting:
```
␛[0;32mI (6566) STOCKLY: Connecting to AWS IoT Core at a13s8kmdddvqm1-ats.iot.us-west-2.amazonaws.com:8883␛[0m
␛[0;32mI (6616) aws_iot: Attempting to use device certificate from ATECC608␛[0m
␛[0;31mE (7666) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700␛[0m
␛[0;31mE (7666) aws_iot:     Unable to verify the server's certificate. ␛[0m
␛[0;31mE (7666) STOCKLY: Error(-4) connecting to a13s8kmdddvqm1-ats.iot.us-west-2.amazonaws.com:8883␛[0m
```
`main.c`: https://github.com/Raunak-Singh-Inventor/Stockly/blob/c250a4fc9c2d20f53c8c7a42fe9bb90174529120/edukit_deployment/main/main.c#L88

rashedtalukder, 2022-05-16:
Appears to be something wrong with the root CA. That's actually a separately attached file that should be included in the certs directory.

Raunak-Singh-Inventor, 2022-05-16:
@rashedtalukder I have already included the `aws-root-ca.pem` file in main/certs. Could you give some pointers as how I could fix the root CA as it shows as `active` on the AWS console.

Raunak-Singh-Inventor, 2022-05-27:
@rashedtalukder Any updates as to why the certificate is not valid? I really want to complete my project.

Raunak-Singh-Inventor, 2022-08-06:
Hi Rasheed,

I am still waiting on your response?

rashedtalukder, 2022-08-08:
Hey Raunak, I’m no longer with amazon as of a few weeks ago and don’t have access to some of the tools that would let me better help you. I’m hoping that someone from AWS will help you. If they don’t, let me know and I’ll schedule some time with you to debug it in a video call.

Raunak-Singh-Inventor, 2022-08-08:
@rashedtalukder Thanks for the reply.

Could you please drop me an email with your new contact id.

closed 2022-08-08 by Raunak-Singh-Inventor