m5stack/Core2-for-AWS-IoT-Kit #97: Blinky Hello World does not work for Apple Silicon devices

goose341, 2022-09-14:
You cannot register the EduKit device with IoT Core on Apple Silicon devices due to an issue with Python dependancy _`cryptoauthlib`_. Per [this issue](https://github.com/MicrochipTech/cryptoauthlib/issues/297) - the published ARM release is actually for x86. 

Attempting to run `pio run -e core2foraws-device_reg -t register_thing` Apple Silicon users receive the following error:

 `File "/Users/usr/.platformio/penv/lib/python3.10/site-packages/cryptoauthlib/library.py", line 105:
    raise LibraryLoadError('Unable to find cryptoauthlib. You may need to reinstall')`

Manually installed `cryptoauthlib` and importing it into Python breaks down the issue to incompatibility with `libcryptoauth.dylib`

`OSError: dlopen(/opt/homebrew/lib/python3.10/site-packages/cryptoauthlib/libcryptoauth.dylib, 0x0006): tried: '/opt/homebrew/lib/python3.10/site-packages/cryptoauthlib/libcryptoauth.dylib' (mach-o file, but is an incompatible architecture (have (x86_64), need (arm64e)))`

darko-mesaros, 2022-09-15:
I can confirm this. It fails on M1 Macs, but it works on x86 devices. I've also tried to use an older version of the package, but that did not work as well.

bryan-hunt, 2022-11-05:
This should have been originally resolved in cryptoauthlib==20221018

goose341, 2022-11-07:
Tested and validated this is now working with `cryptoauthlib==20221104`

closed 2022-11-07 by goose341